opentx/opentx #7142: Horus X10S goes to safe mode while changing model

flyingdockp, 2019-12-01:
I have a FrSky Horus X10S (and the X10S Express) running OpenTx 2.3.1
I have made several model categories, each containing several models.
When I change too fast or I walk through too many categories, the software hangs and the receiver reboots in safe mode.

ALso, as pointed out in another issue, which is now marked as milestone, the Horus does not remember which model I used before the restart. 

Total number of categories: 
[Horus 10XS 2.3 modellenlijst nov 2019.zip](https://github.com/opentx/opentx/files/3908672/Horus.10XS.2.3.modellenlijst.nov.2019.zip)

3djc, 2019-12-01:
Have you tried with another sd card ?

flyingdockp, 2019-12-02:
Yes, two transmitters (Horus x10s and Horus x10s express), 2 sd cards same problem

robustini, 2019-12-03:
Have you tried with the nightbuilds?

closed 2020-05-31 by kilrah