opentx/opentx #7150: Direct telemetry through port uart3 in Jumper T16

Guillesan, 2019-12-04:
It would be extraordinary if you could use the uart3 of the Jumper T16 board. Although this already works by bluetooh, as you know the connection by bluetooh is quite short distance, I wish I could use this port to connect for example a wifi module like the one I am using in my Taranis 9XE. I enclose photos of what I try to explain.
[image]

3djc, 2019-12-05:
Since we never heard of wifi for any OpenTX transmiter, really hard to tell. Any link to what you have on your X9E ?

Guillesan, 2019-12-05:
Hi, the explanation of using wifi is a bit long but I will try to summarize it. Together with several colleagues we started a project called www.u360gts.com, its purpose is to mount a multi-platform tracker antenna (telemetry), this is very advanced, you can see complete information in that direction. Initially telemetry is forwarded to the tracker antenna through the bluetooh modules of the Taranis 9xd, 9Xe, Horus 10 and now thanks to 3djc in the Jumper T16. But the bluetooh connection is weak in terms of distance, so we looked for a solution and it was to use Wi-Fi modules https://www.banggood.com/Geekcreit-DT-06-Wireless-WiFi-Serial-Port-Transparent- Transmission-Module-TTL-To-WiFi-Compatible-With-Bluetooth-HC-06-Interface-ESP-M2-p-1141047.html? Rmmds = myorder & cur_warehouse = CN
These modules behave the same as the hc-05/06 creating a transparent serial connection normally at 115200 and connecting the Taranis x with the tracker or pc antenna (Mission Planner) more robustly and greatly extending the connection distance. For use they connect to the serial port on the motherboards of the taranis, attached photo of my connection in my Taranis 9xe. Knowing that the Jumper T16 has the call uart3 on board would be interested could also be used to extract the telemetry received by that port and connect it to the outside by the wifi modules described above. I do not know if it is possible but I create a series of great interest for the users of this station. Sorry for the extension of this answer, I am available to answer any questions. a greeting
[image]
This is the modification in the Taranis 9XE, you can see the wifi module connected to the output of the uart.
[image]
A photo of the tracker antenna receiving its own CRSF telemetry from TBS Crossfire directly extracted
[image]
[image]
Thank you for your attention, Regards

3djc, 2019-12-05:
There is not free uart on T16, the best you could do is try to reuse BT uart which is uart6

Guillesan, 2019-12-05:
Thanks for your quick reply. A pity is not free uart. I had thought the same usart the rx-tx input of the interior bluetooh module in the Jumper T16. I will test and inform if it is of interest.

redjacketKR, 2020-08-25:
So how was the result going on Guillesan?

brainbubblersbest, 2020-12-15:
#8073

closed 2021-09-22 by stale[bot]