opentx/opentx #7151: Internal Multi Radio GUI does not show Proto Sync Status line using Jumper T16

MRC3742, 2019-12-05:
Using either the older X9D+ or Jumper T16 Radio GUI, on the External Multi, the display shows Proto Sync Status information right below the Module Status line of 1.3.0.47 AETR.
Switching to the Internal Multi Module on the Jumper T16 this information line does not display.

Both the External and Internal Modules were compiled using the same base config. with only "#define INVERT_TELEMETRY" commented  for building Internal Module firmware.

**General information**
 - OpenTX version: 2.3.3RC1
 - Hardware: Jumper T16

3djc, 2019-12-05:
We need @pascallanger to confirm, but I think he removed MULTI_SYNC from default firmware config as it seem to produce issues.

On 2.3.3 this line will only show if sync is actually enabled

pascallanger, 2019-12-05:
MULTI_SYNC has never been activated in releases since it's still a work in progress.
However, it was at some point by default in my dev "sport_send" branch. If you are still using this dev branch code I strongly advised to upgrade your modules with the main branch.

3djc, 2019-12-05:
So I'm taking it that there doesn't seem to be an issue in OpenTX, so closing this

closed 2019-12-05 by 3djc