opentx/opentx #7153: T16 Pro RC1 DSM sub protocol missing on internal module.

fabioeh, 2019-12-05:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

3djc, 2019-12-05:
[image]

Working correctly. Make sure your module version is recent enough

fabioeh, 2019-12-05:
check my screen shot

3djc, 2019-12-05:
Please post your otx, definitively not happening here

fabioeh, 2019-12-05:
are you on .47

3djc, 2019-12-05:
[image]

Same, working here

fabioeh, 2019-12-05:
[Fabio-T16.otx.zip](https://github.com/opentx/opentx/files/3926766/Fabio-T16.otx.zip)

fabioeh, 2019-12-05:
only happens on the internal the external works.

3djc, 2019-12-05:
only when external is set as RX actually

fabioeh, 2019-12-05:
Looks like you were able to reproduce the problem.
Thanks for looking in to it.

3djc, 2019-12-05:
The fix should be in tomorows nightly

3djc, 2019-12-05:
Will have to wait one more nightly, sorry

closed 2019-12-06 by bsongis

fabioeh, 2019-12-07:
Thank you, Just tested n63 and the problem is resolved.

commit 264c479d (2019-12-05): Fix subprotocol sometime not shown
This fixes #7153
PR #7147 (2019-12-06): Use PROTOCOL_TELEMETRY_MULTIMODULE for internal multi when available
This fixes TRSS TLQI not been found with internal multi

This fixes #7149
This fixes #7153
commit fdc154c5 (2019-12-06): Use PROTOCOL_TELEMETRY_MULTIMODULE for internal multi when available (#7147)

* Use PROTOCOL_TELEMETRY_MULTIMODULE for internal multi when available

* Fix subprotocol sometime not shown
This fixes #7153