opentx/opentx #7169: Bluetooh module output power configuration

Guillesan, 2019-12-08:
Days ago I am testing the bluetooh function on the Jumper T16, telemetry sending works correctly. However I have seen that the connection distance between the bluetooh module and the outside, in my case a tracker antenna is really minimum at most about 2 or 3 meters. Heen specifically installed an outdoor antenna to maximize this distance without positive results. The module used is the cc2540 (bluetooh) that has an AT command to configure the transmission power.
AT + POWE? OK + Get: [P1] None
  AT + POWE [P1] OK + Set: [P1] P1: 0 ~ 3
0: -23dbm
1: -6dbm
2: 0dbm
3: 6dbm
Default: 2
Would it be possible from opentx to configure the output to 6dbm?
I think that the default lamp power is 0dbm, very little, it would be nice to upload it to 6 dbm since the module allows it.
a greeting

3djc, 2019-12-08:
You should have a look at your hardware. Just did a test at 12m without issue between an x7 and a T16, and I could not try longer distance because its pouring outside

robustini, 2019-12-10:
@Guillesan If you want to try I think it's enough to edit the /radio/src/bluetooth.cpp file to give that AT command during initialization.

Guillesan, 2019-12-10:
I'm sorry, despite myself I don't have the ability to program this.  if it were possible to put it another way?

closed 2020-09-05 by stale[bot]