opentx/opentx #7171: [OpenTX V2.2.3] - Flash bootloader from SYS folder causes crash on Horus X10S Express

robustini, 2019-12-09:
Flashing the last bootloader from the SYS folder menu sends the Horus X10S Express in "Emergency Mode", even if the bootloader writes it regularly.

 - OpenTX 2.3.3 (self compiled, but with RC1 is the same)
 - Hardware Horus X10S Express

3djc, 2019-12-09:
Do you mean the last line of directory ? or bootloader of RC1 ?

robustini, 2019-12-09:
I mean flashing the bootloader from the file menu of the tx.

3djc, 2019-12-09:
Is it specific to x10 express ? I have tried x10 and t16, but cannot reproduce

robustini, 2019-12-09:
Specific (i think) with X10S Express, see the video, i've no issue with my T16:

https://youtu.be/un3RZAajP2Q

3djc, 2019-12-09:
Then I can't help, will try to find someone who can

closed 2019-12-10 by bsongis

bsongis, 2019-12-10:
Reproduced and fixed

commit 97c56343 (2019-12-10): Fixes #7171