opentx/opentx #7174: OpenTX build instructions using QT not working for 2.3

lshems, 2019-12-09:
I do get to clone the 2.3 default branche locally, compile it with 'release' 'companion'.

it compiles, but the DLL files to run the resulting companion.exe seem to fail.

Any help appréciated

MRC3742, 2019-12-09:
This is the same issue I believe that I reported on Issue #6831

If you first install one of the Companion 2.3 builds and then copy in (over) the files that you get from the compile build it will then work properly.

Hope this helps - Rich

lshems, 2019-12-09:
thanks. I will check.

zyren, 2019-12-11:
You need install python-clang

MRC3742, 2019-12-11:
python-clang binding is only needed to build firmware as discussed in #7176 

Rich

lshems, 2019-12-11:
I tried to copy over a clean nightly install. No good, still missing DLL.

It is rather frustrating that after five years of spending several hours everytime I want to contribute something, I get blocked at not being able to compile.

It is really not inviting anyone to get into this, when you want to get started with no experience on something small.

I just want to copy some functionality already in the mix lua api to the inputs lua api. Seems doable for me.

But not without a working compile possibility?

Please try to put some effort in getting this sorted out.

dwheel, 2019-12-19:
I am building with MinGW 32 bit. In the build settings, it points to where the dll files are located. In my case, they are in C:\Qt\5.12.2\mingw73_32\bin. Run companion.exe and when it reports the dlls missing, go to this folder and copy then over to the companion.exe build folder. 

It would be nice if they were automatically copied over, but since they are not, this is a work around. 

Here is a print screen showing where to look.

[image]

lshems, 2019-12-20:
Thanks a lot for helping out a noob. this was what is missing.

I can now see if I can add some work.

:)

closed 2019-12-20 by lshems