opentx/opentx #7175: Para wireless bluetooth module only

mortehoe, 2019-12-09:
I have installed the para bluetooth module on Frsky horus X10s together with ISRM module.Wireless module which worked properly when installed with opentx 2.3.2 'internalaccess' enabled.I have since removed ISRM module (too many issues) and installed opentx without 'internalaccess' function.
Since then the bluetooth module appears to operate with less resolution (seems to step slower when servos deflecting) in bluetooth trainer mode linking with xlite pro.

3djc, 2019-12-10:
We reviewed the code and could not find any reason why removing internalaccess would change in any way BT operations

kilrah, 2019-12-10:
Probably just RF interference from the internal module causing it to drop packets...

mortehoe, 2019-12-12:
[image]
I have replaced the ISRM module (internalacces)and it does not seem related to the module the only thing updated since last test on 2.3 nightlies is now to 2.3.3 rc1.
This exclamation appears on initial setup of bluetooth trainer mode and I have to clear and discover again before reconnection (also after powering off/on it can take a couple of minutes to connect again.
[image]
when discovering devices (on master) at first it showed both devices addresess then it only showed one as pictured above after having to re-discover (correct address)
I am fairly sure when I tested originally when  the issue of bluetooth not working (#6910) was fixed it was working fine in nightlies with no loss of resolution.

closed 2020-05-19 by kilrah