opentx/opentx #7176: [Compiling 2.3 using Qt Creator] Problems causing errors and unsuccesful builds

MRC3742, 2019-12-09:
I am currently having a few problems using Qt Creator to compile OTX 2.3.
I have been building as per Wiki instructions successfully for about a year.
This started happening somewhere after or around commit #7098
I now had to add clang to python (command "python -m pip install clang") and add libclang.dll somewhere on my Windows Path to be able to compile for the T16 without it erroring out.

Even after adding the above requirements, now using build steps "all" or "install" these currently do not build the firmware file.
You need to add an additional build step for "firmware" in order to have it compile.
Are there additional requirements not yet listed in the Wiki?

Is Qt Creator still being supported for 2.3 builds? 
Or is Docker the current preferred method.
Will Docker build Companion, Simulator and Firmware?

**General information**
 - OpenTX: 2.3.3 from current master build (2.3)
 - Hardware Jumper T16

Thanks - Rich

kilrah, 2019-12-10:
> Even after adding the above requirements, now using build steps "all" or "install" these currently do not build the firmware file.

That's intentional.

MRC3742, 2019-12-10:
Then it appears that it is still building as is expected!
I didn't see any mention of it in the current commit comments.

Are the other requirements I have listed also now necessary and correct?
If so I could make some additions to the Wiki on my repository for review?

Thanks - Rich

kilrah, 2019-12-10:
clang and the python binding should only be needed when building firmware, but yes they were a new dependency added for those by the PR you linked to. 

Firmware target was removed from "all" shortly after precisely becasue it's unnecessary when building the PC package, saves time and the need for adding the new dependency to the build servers.

MRC3742, 2019-12-10:
Thanks for the explanation.
I uninstalled the clang to python binding and the "all" target builds correctly now but "firmware" errors out as you described it would. Reinstalling the clang binding and now it's back to building "firmware" again.

I'll close this issue as "all" and "install" appears to be working properly as you have changed the expected build process to exclude the firmware file from being built in those build steps.
Just the Wiki needs to be updated for added dependencies for build versions of 2.3 firmware.

Thanks Again - Rich

closed 2019-12-10 by MRC3742