opentx/opentx #7179: Momentary Switch SH slow to respond - Horus X10s 2.3.3RC1

aussiepage, 2019-12-10:
Switch SH slow to respond - only happens on Horus

To reproduce:-
1. Setup an unused mix to with source SH
2. View channels using monitors page.
3. Toggle momentary switch several times.

Can take several seconds to respond to switch change.

A clear and concise description of what the bug is.

**General information**
 - OpenTX version 2.3.3 RC1
 - Hardware Horus X10S

3djc, 2019-12-10:
I cannot think about anything else but a hardware fault

aussiepage, 2019-12-10:
Hi - issue was repeated on a friends Horus running same version of
software.

3djc, 2019-12-10:
Just tried here and I cannot replicate. Have you tried in switch test screen as suggested?

aussiepage, 2019-12-10:
Thanks for help. I investigated further and found that I also had SH mapped
to Screenshot in global functions. Removing this mapping fixed issue.

Probably not an issue that would affect many as screenshot function not
used often.

Thanks

closed 2019-12-10 by kilrah

bsongis, 2019-12-11:
would you please post your setup? what is linked to SH?

kilrah, 2019-12-11:
It's just the UI being frozen during the screenshot and the channel monitor thus not showing the change, a servo connected to the channel would actually move

bsongis, 2019-12-11:
ok, normal then