opentx/opentx #7181: [OpenTX V2.3.3

robustini, 2019-12-10:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

robustini, 2019-12-10:
I don't understand, in Companion the Multi-Trainer now is available with T16, but not on Horus target like X10S and X12S... what are you doing?

kilrah, 2019-12-10:
Mult trainer can't be use on those since the smart port is used by the internal module.

robustini, 2019-12-10:
How can i not use it?
I've been using it for a month!

kilrah, 2019-12-10:
What's your config exactly?

robustini, 2019-12-10:
Horus X10S Express ACCESS and Multiprotocol module.
With ISRM the smart port pin is not used by the internal module if it removes a pin from the module.
@raphaelcoeffic did all this work unnecessarily?
I don't understand why you removed this option, I've been using it for months.

kilrah, 2019-12-10:
Yes, ACCESS upgrade AND hardware mod it works, but people who do that will be counted on one hand's fingers.

Removal was likely a side effect of cleaning up the choices. 

We have to decide what we do with this kind of stuff, and it's not easy. If it's there without the proper hardware configuration people will try to use it and complain it doesn't work. If it's not there at all then people who do specific mods are missing choices. But we can't have options for every single possible modification either.

robustini, 2019-12-10:
This is understandable, but removing the possibility seems to me quite "dictatorial".
We worked on it for months, it was inserted after countless tests in branch 2.3, then approved, and now it is taken off?
It does not seem ethically correct.
At least leave the possibility to enable it by compiling the firmware with a cmake option, eliminating it seems incorrect.
I ask you, @bsongis and the other lead developers to reconsider.
I'm collaborating with your code, making continuous adjustments in the Italian language because in 50% of my use the Multi-Trainer is fundamental, not having it for me is deleterious.

robustini, 2019-12-10:
I saw where it was deactivated, I recompile it on my fork as it was before, thanks anyway.

kilrah, 2019-12-10:
> We worked on it for months, it was inserted after countless tests in branch 2.3, then approved, and now it is taken off?

It was always developed with the T16 as the target, the whole effort was precisely triggered by the discovery that it had independent connections and that it would thus allow for this...

robustini, 2019-12-10:
No problem, I've already created a batch here that automatically enables it in the code before compiling it.

bsongis, 2019-12-11:
I reopen until we find a suitable solution.

robustini, 2019-12-11:
Thanks @bsongis , I really appreciate it!
But I also understand what @kilrah writes, I understand it and I agree with it, unfortunately the average user sees this option that can be activated, he tries to enable it and use it, noting that it doesn't work and he starts to open issue here or to write posts on the forums asking why, it would be deleterious.
Given that it is something that requires a hardware modification on the ISRM module or the pcb that houses the module, even if very simple, this thing should be well explained and it is not easy to make it public, and many will ask for explanations of the non-functioning.
I think the best solution is to leave it enabled only for those who know how to compile the code, so if it is doing so it means that it is also aware of the hardware modification.
The problem is that Companion should also be compiled, and it becomes quite boring.
I honestly don't know what the solution could be.

robustini, 2019-12-11:
The only possible solution maybe in my opinion could be to insert "multi-trainer" in the options that can be selected in Companion, and in the relative popup write that if not used with Jumper T16 requires ISRM with a hardware modification, with IXJT it wouldn't work anyway.

3djc, 2019-12-11:
- People don't read
- Companion option are a **major** source of issues and we are trying our best to decrease their number

robustini, 2019-12-11:
@3djc you are absolutely right, but we have included "internalaccess" for the X10 and X12S in Companion, and that does not require a hardware modification? :-)
If there is that there may also be this.

robustini, 2019-12-11:
However, something different between X10S Express ACCESS and X12S with ISRM is there regarding the multi-trainer.
As you can see from the video (listen to the audio) the sport telemetry of the ISRM module stops working when I enter the MODEL menu, something that never happened to me with the X10S Express and the modification of the module I posted above.
In the same way I'm not able to flash the multiprotocol module, I'm forced to disable both ISRM and himself to do it.
The thing is strange because the X12S with the ISRM and the hardware modification should behave theoretically like the X10S Express.
For this thing I ask for an answer from @raphaelcoeffic, who may perhaps know why, honestly I don't understand where the difference is, since the ISRM does not use the sport pin for telemetry.

https://youtu.be/oGhtD3iEZd4

robustini, 2019-12-11:
Could it be that with "internalacces" enabled there is something that is still used by ISRM sport pin, maybe something want disabled in case of active internalaccess?

robustini, 2019-12-11:
I found a problem, in the internal module selection iIXJT also appears, but if I chose the "internalaccess" option for ISRM in the firmware this option "IXJT" should not be selectable.

fabioeh, 2019-12-11:
If cutting a pin let’s me use Master/multi on my x10 with access upgrade. Please add it to OpenTX

Thanks

robustini, 2019-12-11:
> 
> 
> If cutting a pin let’s me use Master/multi on my x10 with access upgrade. Please add it to OpenTX
> 
> Thanks

Absolutely yes!

fabioeh, 2019-12-11:
You could always replace it with one that has the pin for whatever reason you need it.

robustini, 2019-12-11:
There is no reason for this pin to exist, it only serves to block the possibility of using both modules at the same time, this apparently was annoying FrSky, perhaps because the average user could make disasters with link losses due RF interference or other thing.

3djc, 2019-12-11:
It is even more complicated than you think. R9M and XJT module do work at same time despite that pin, so it seems selective in what it is blocking

fabioeh, 2019-12-11:
If there is no reason for it just let OpenTX users remove the pin and enjoy helping someone at the field by connect there transmitter to the X10 multi and enjoy their day.

robustini, 2019-12-12:
> 
> 
> It is even more complicated than you think. R9M or external XJT module do work at same time as ACCESS internal despite that pin, so it seems selective in what it is blocking

And it is no coincidence, R9M is made by FrSky! :-)
Who knows why with Crossfiere it is not so, by pure chance? :p

fabioeh, 2019-12-20:
Looks like it’s a win for the completion 2.3.3 released with this feature disabled on the x10
More power to that others radio.

closed 2019-12-23 by bsongis

gretel, 2020-03-06:
got `x9lite` and a multimodule compatible with https://github.com/pascallanger/DIY-Multiprotocol-TX-Module - can i make a hardware hack on this combo to be able to use `ALLOW_TRAINER_MULTI` ? regards

PR #7226 (2019-12-23): Introduce ALLOW_TRAINER_MULTI 
Provide a manual compilation option to force TRAINER multi  for Horus with hacked hardware

This closes #7181