opentx/opentx #7184: Pot or Slider as Throttle Idle Cut

rosnik, 2019-12-11:
Hi guys,
this thideo show hot to use instead trim one pot or one slider:
https://youtu.be/WoJdmc3vZlc
there are few mode to do it, like suggested on the rcgrop tread:
https://www.rcgroups.com/forums/showpost.php?p=43346471&postcount=7431
and 
https://www.rcgroups.com/forums/showpost.php?p=43345161&postcount=7429
FEATURE REQUEST :
May Open TX will offer in IMPUT, THROTTLE and trim; 
 add SLIDERS and POTENTIOMETER like trim chose options??

With this unfortunately not present feature,
 it will ask for only 5 seconds to do it.
See video from 3:15" to 3:49" about.

thank you
Nico

3djc, 2019-12-11:
There are plenty of way to do this with existing OpenTX toolset.As you mention it, everyone as it own view of how it needs to work, so build your own, that’s the whole point of opentx

rosnik, 2019-12-11:
Thank  you.
you can freely close

Eldenroot, 2019-12-15:
It is but why not to make it easier?

closed 2019-12-15 by 3djc

kilrah, 2019-12-15:
Because it would require huge changes in OpenTX for something people can already do easily.

3djc, 2019-12-15:
And "everyone as it own view of how it needs to work"