opentx/opentx #7190: OTX 2.3.x: trim beeps skipped when trim held

RC-SOAR, 2019-12-14:
Hardware: Jumper T16 (OTX 2.3.3 RC1), Horus X12S.  

**Expected behaviour**: transmitter should issue a beep with each trim increment. 
**Actual behavoiur**: when the trim is held, some beeps are skipped. Applies to all trims, regardless of direction.

I've seen this on both the T16 and X12S on various 2.3 nightlies and 2.2.3 RC1. Video below illustrates the issue in a new model on the T16:

https://youtu.be/ujpX2_9XQYU

(Works as expected on X9D running 2.2.3).

3djc, 2019-12-14:
That’s because of the audio been muted to prevent the continuous background noise

RC-SOAR, 2019-12-14:
Point taken, but is there any way this can be fixed? Hate to pester given the great work you've put into 2.2.3, but IMO accurate real time feedback on trim action is essential and this is  definitely a step back in terms of usability. I thought this might just be an issue with the nightlies hence the delay in reporting.

Eldenroot, 2019-12-15:
There should be a beep when you reach the center position...

3djc, 2019-12-15:
There is an audio message when thats the case

kilrah, 2019-12-15:
And a beep with a different tone if the voice file is missing.

PR #7193 (2019-12-16): T16 mute improve
This fixes #7190
closed 2019-12-16 by 3djc