opentx/opentx #7192: Bluetooth QX7S to android freelink

bangor-flyer, 2019-12-14:
Hi running the latest firmware on QX7s. Using frsky sport airlink s I can read all telemetry sensors using the freelink app using bluetooth on android. 
Can connect and read all telemetry sensors through RX and display on QX7s but cannot seem to broadcast to freelink app from TX using bluetooth . have changed bluetooth setting to telemetry on Tx.

williajgr, 2020-01-17:
having the same problem with my taranis se 2019 and horus x12s. cannot seem to broadcast to freelink app from TX using blue tooth . have changed blue tooth setting to telemetry on Tx. I can connect to the blue tooth on the radio but no sensor data to the monitor.

pulquero, 2020-09-11:
I've tested Free Link on a Samsung S10 with X9D SE 2019 and the Monitor picks up telemetry, albeit the update rate is slow. Additionally, I've also tested X9D against various GATT clients, and Free Link against various GATT servers.

I can offer the following observations which may help address the various bluetooth issues. When I tried Free Link against a server that sends a **single** SPort packet per notification, the Monitor updates at a more reasonable rate. The important point here is by default the MTU is 23 bytes, which leaves 20 for data. A single SPort packet fits comfortable. The opentx code however tries to send two packets (https://github.com/opentx/opentx/blob/2.3/radio/src/bluetooth.cpp#L278). For this to work (and it seems to be the case), the TX bluetooth module must be using an MTU >= 25. Depending on various versions of android, etc, I suspect some assume it is 23 bytes and truncate the data, leaving partially corrupted packets (which could lead to the the slow update rate observed).

I would therefore suggest that the simplest thing to do to ensure a high level of interoperability is to change the opentx code to just send single SPort packets per notification. Even better would be to adjust the number of SPort packets per notification based on the **requested MTU** from the **client**. The server can't dictate the MTU to the client. When I try requesting the MTU is set to 25 or higher from the client side, the android stack seems happy to accept the two SPort packets per notification that opentx is trying to send out by brute force.

Hope this is of some assistance.

closed 2021-06-18 by stale[bot]