opentx/opentx #7194: Startup (power on) menu to select model - enhancement

Julien31fr, 2019-12-15:
Context :
When I start my remote, I always need to plug my 2 antennas, even if i just want to do simulator via USB, where the radio emission is disabled, because I don't remember the last model I have used. Turning on a radio emission without antenna could cause damages.

Solution 1 :
At startup (power on), instead of having the last model to be charged, we could have a menu that allow to choose the model to be started, before any radio emission.

Solution 2 : 
At startup (power on), we could have the last model to be loaded, with a grace period of 10 seconds before any radio emission to start, to have the time to change to the desired model.

Additional context:
I have asked on the french facebook group "Culture Fpv", and I'm not the only one with this issue.

I use an Xlite that has :
- internal antenna,
- external antenna T Shape, because the internal antenna is not good enough,
- external module R9 lite, with super 8 antenna.

I use models in those situations :
- USB (joystick, without radio emission) : to do simulator
- External antenna T Shape to use RXSR modules
- External R9 lite module + Super 8 antenna

Thanks for your great job.

Eldenroot, 2019-12-15:
Great idea!

OpenUAS, 2020-01-06:
@Julien31fr Your wish is understandable, scenarios like this are why e.g. in the past Graupner TX's when plugged in a trainer cable switched on the TX without switching on the RF internal circuit. This originally  to prevent accidental channel switch on in the field when crystals where still used to determine which channel. 

However the solutions 1 and 2 proposed are not ideal. Better if possible would be if USB is plugged in, and option to disable RF output on all modules if HID device is detected. or anything that doesn not involve quick TX start-up pollution ;)

Maybe as an interim solution for you is an option to use a Wireless USB RX HID module, they do exist.  Search for "_URUAV Wireless Adapter for USB Dongle with Simulator Supports FrSky Flysky RadioLink Walkera SBUS PPM Receiver_". Or maybe use a small dummyload on your antenna output. Or swap a model just before you switch of your transmitter, then you don't need remember the last model you used. ;). Then if you switch it on, the Sim model is chosen...

i3dm, 2020-01-13:
actually it would be good to offer a "RF OFF" state on powerup if chosen.
Futaba does allow this when switching on with throttle above idle. after disabling the throttle alert, it asks you if you want to run RF ON or RF OFF.
this is very useful when doing setup work on the Tx without your model. would be a great feature to add to OpenTX.

borsti2000, 2020-08-17:
Hello, I think a function to enable a "Model selection on Startup" is very usefull. 
I use different Models if I on a exposition. And it would be nice to have the possibility on every startup to select the model which I will actualy use. Also for safety reasions. To have less posibility to have the wrong model selected.
If I am right, I could also create an "empty" model and select them as "default", but If the model selection is on startup it is more comfortable.
For sure, for users who dont want a selection, this option should have the possibility to deactivate in setup.

3djc, 2020-08-17:
Safety needs to happen with modelId/rxnum,  not this, as you can easily pick the wrong one

i3dm, 2020-08-17:
but how about my original request? it really is super useful not to have the radio transmitting when setting up a model on the Tx.

borsti2000, 2020-08-17:
I am not sure what you mean with "modelId/rxnum". For sure, thes selection on startup should be the same as in the "Model selection screen" to see that you will select the right one.
And in my ponit of view it is a "part of safety" because you have select one (one additional step) instead to have the "wrong" automaticaly activated.

kilrah, 2020-08-17:
The model match / receiver lock / whatever each manufacturer calls it, if the wrong model is selected then it just won't respond.

borsti2000, 2020-08-17:
If each model have a different manufacturer or RTF models, maybe. But not if you have mounted receiver and an external 2.4ghz module witch get only the PPM (e.g. Jeti duplex).
In this case the Model will connect also if the "wrong" model is selected.

i3dm, 2021-02-14:
any news about this feature guys? ;)

i3dm, 2021-08-21:
so nobody thinks this feature is important?