opentx/opentx #7195: OTX 2.3.3-RC2  6P position indicator not working on T16 display

BatesKG, 2019-12-15:
Really minor but I thought I would let you know.  I just noticed it after moving from JumperTX to OpenTX 2.3.3-RC2.    When you turn the slider indicators on for the TX screen, the indicator at the bottom center that shows what position the 6P switch is in always indicates position #1 regardless of what the 6P switch is selected to.  In the analog debug screen or if you assign a channel to 6P in Mixers, the value is still changing when selecting 1-6. It's just the indicator that doesn't seem to be working.  You can see this behavior in the Companion radio simulator as well.
 The indicator did change when running JumperTX.

  Thanks for all the great work you are doing!

Kevin

3djc, 2019-12-15:
Have you selected the 6pos while doing the calibration ?

BatesKG, 2019-12-15:
No….I missed that!    Doh!      I’ll close my bug report.    
That was a SUPER fast reply by the way!   Wow!  :)
 Thanks again.

closed 2019-12-15 by BatesKG