opentx/opentx #7199: TH9x/AR9x 2.3.3RC2 Switch Detection Errors

OldGazer, 2019-12-16:
I was messing around with my TH9x yesterday and while I was setting up a new model I discovered a strange issue. 
The initial 4 channel setup is fine, but when I go to setup Channel 5 (Aux1) with ANY SWITCH, the switch is not detected properly like this:

TH9x Switch Detection
1. ELE is detected as RUD.
2 RUD is detected as THR
3. THR is detected as 3POS
4. AIL is detected as ELE
5. GEA is detected as AIL
6. TRN is detected as GEA
7. 3POS is detected as Ail (there is a little box graphic in front of the letters ???)

If I try to select 3POS manually by scrolling through the list of inputs, the 3POS does not work, THR does.

Sticks and Pots are detected properly.

On the model display screen all switch, pot and stick, and trim graphics operate properly.

Using Mission Planner > Initial Setup > Mandatory Hardware > Radio Calibration I can verify that when I assign 3POS to Channel 5 (even though it is detected as Ail), the switch does in fact change Pixhawk flight modes.

OldGazer, 2019-12-16:
I can also reproduce this issue in the firmware simulator...

kilrah, 2019-12-16:
I think the names are shifted

OldGazer, 2019-12-16:
Good possibility.  Its always the little things that jump up and bite you...

closed 2020-06-05 by dgatf