opentx/opentx #7200: Add the Name of the Timer in model.getTimer()

Storchy1, 2019-12-16:
Is it possible to expand the table of the timer function?
In the struct TimerData, i think the name of the timer is already included.

PACK(struct TimerData {
  int32_t  mode:9;            // timer trigger source -> off, abs, stk, stk%, sw/!sw, !m_sw/!m_sw
  uint32_t start:23;
  int32_t  value:24;
  uint32_t countdownBeep:2;
  uint32_t minuteBeep:1;
  uint32_t persistent:2;
  int32_t  countdownStart:2;
  uint32_t direction:1;
  NOBACKUP(char name[LEN_TIMER_NAME]);
});

so perhaps it is possible to "lua_pushtablestring" the name in this function
static int luaModelGetTimer(lua_State *L)
{
  unsigned int idx = luaL_checkunsigned(L, 1);
  if (idx < MAX_TIMERS) {
    TimerData & timer = g_model.timers[idx];
    lua_newtable(L);
    lua_pushtableinteger(L, "mode", timer.mode);
    lua_pushtableinteger(L, "start", timer.start);
    lua_pushtableinteger(L, "value", timersStates[idx].val);
    lua_pushtableinteger(L, "countdownBeep", timer.countdownBeep);
    lua_pushtableboolean(L, "minuteBeep", timer.minuteBeep);
    lua_pushtableinteger(L, "persistent", timer.persistent);
  }
  else {
    lua_pushnil(L);
  }
  return 1;
}

3djc, 2019-12-16:
What exactly do you want ??? model.getTimer(timer) to return also the name of the timer ?

Storchy1, 2019-12-16:
[image]
I want the timer Name "Flugzeit" and "Total"

Storchy1, 2019-12-16:
Thanks a lot for processing and adding to 2.3.4 of my request

PR #7358 (2020-02-06): Add name handling to LUA timer functions
This fixes #7200
closed 2020-02-06 by bsongis