opentx/opentx #7213: No data from `FrSky SxR.lua` for S6R

rhass, 2019-12-18:
**Describe the bug**
The receiver I am using is an S6R with the latest firmware flashed to the receiver.
When executing `SCRIPTS/TOOLS/FrSky SxR.lua` (v2.00) on an X9D+ (non-2019 version) the script does not populate any of the fields in either of the two setup screens. This appears to affect both 2.3.3RC1 and 2.3.3RC2.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to **SD CARD**
1. Scroll down to `SCRIPTS`
1. Scroll down to `TOOLS`
1. Scroll down to `SCRIPTS`
1. Long press `FrSky SxR.lua` and select `EXECUTE`
1. The values will remain as `---` for both menus after it appears to scan the receiver.

**Expected behavior**
The values configured on the receiver should appear and become configurable.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version: 2.3.3RC2 (20cc9f9c)
 - Hardware: Frsky X9D+ (pre-2019 version)
 - OpenTX Companion: 2.3.2, Nov 16 2019 (Windows)

rhass, 2019-12-18:
I just discovered this seems to happen when I have both a high precision vario and a FLVSS connected to the s.port.  After disconnecting the sensors, I was able to configure the receiver. However after reconnecting the sensors,  I can see some of the settings, but some values don't seem to appear while these sensors are connected.

3djc, 2019-12-18:
Do you have enough distance between the radio and the sxr ?  Swamping can alos cause data to be missed

rhass, 2019-12-21:
I do have enough distance. I started having issues with sensors not being detected and then reverted the firmware back to 2.3.2. After doing so, all my issues seem to be resolved. Is suspect there is something in the 2.3.3 firmware which is not playing nicely with the SxR receivers.

3djc, 2019-12-21:
That's quite strange because no changes have been done in that area between 2.3.2 and 2.3.3, and some --- and some not indicate RF issue more than anything else

closed 2020-10-12 by stale[bot]