opentx/opentx #7219: Companion OpenTX 2.3.3RC2

Gaylian, 2019-12-20:
No indication that "Write Models and Settings to Radio" is working or finished. (It does write back as when you read models back they have changed).
My Radio is a Jumper T16 Pro

OpenUAS, 2020-01-03:
Dear @Gaylian could you change the issue _title_ to a bit more descriptive one please. Also the _Body_ of your issue could use some TLC in the text to describe your issue a bit better. 

All this will help developers to know what if at all needs to be fixed and in turn more chance that if something is not working will be fixed. TIA

3djc, 2020-01-03:
There is one in your left bottom bar. It display quickly, but is there nontheless

kilrah, 2020-01-03:
Clicking OK everytime on a "written successfully" would be a pain. You get a message if there's an error, if nothing pops up it means it's OK.

closed 2020-01-03 by kilrah