opentx/opentx #7220: Won't go into boot loading mode after updating to 2.3.3

mccartysr, 2019-12-20:
After I updated the bootloader to 2.3.3 it won't go in the bootloader anymore.had go back to bootloader 2.3 to get it to work .

3djc, 2019-12-20:
Sorry but into lend my Cristal ball to Santa Claus so he could better prepare Christmas, so I have nothing left to guess what radio you are using...

mccartysr, 2019-12-20:
Sorry 2019 x9d +se

kilrah, 2019-12-20:
Working fine here

mccartysr, 2019-12-20:
An i was wrong it is bootloader 2.3.1works if i update the bootloader to 2.3.3 it won't go it to bootloader on my 2019 x9d+ se

mccartysr, 2019-12-20:
I have both bootloaders on my SD card.would that make 2.3.3 not work?

3djc, 2019-12-20:
No, the sd card is not even needed or used to enter bootloader

mccartysr, 2019-12-20:
Sorry to waste your time. I put the wrong profile in .i had a bootloader x9d not the 2019 x9d.

closed 2019-12-20 by kilrah

bsongis, 2019-12-20:
Yes. I just checked, everything works perfectly