opentx/opentx #7221: [OpenTX V2.3.3] - Wrong signature warning on X12S when try to flash MPM

robustini, 2019-12-20:
Could you check if this is indeed the case?
On X10S Express and Jumper T16 no problem, when I try on X12S I have this error.

3djc, 2019-12-20:
Could you share the file you are trying to flash ? Fairly strange as the code should be the same

robustini, 2019-12-20:
For the multimodule or the tx?
If you mean the MPM firmware I flash the same file in the others tx and work fine.

robustini, 2019-12-20:
[multi-stm-1.3.0.52_EXT.zip](https://github.com/opentx/opentx/files/3989969/multi-stm-1.3.0.52_EXT.zip)

3djc, 2019-12-20:
For the multi, it will make it easier to replicate. Also is your x12 using internalaccess ?

robustini, 2019-12-20:
Yes, internalaccess enabled.

3djc, 2019-12-20:
Ok i will work on it tomorrow if I can replicate on simu. If not, I’ll need to find someone with x12

3djc, 2019-12-21:
OK, I cannot reproduce, but the wrong signature message seems to indicate the radio does not like the info comming back from the multi module bootloader. Does flashing this very module on another radio work ?

robustini, 2019-12-21:
Yes, with other radio work, with the same module.
I removed that pin from the ISRM module for Multi-Protocol but I don't think that's the problem.

3djc, 2019-12-21:
@raphaelcoeffic your thoughs ?

robustini, 2019-12-21:
If the X12S uses this pin to flash the MPM then I've cut out the possibility of flashing MPM with OpenTX, but on the X10S Express I did the same thing and the flashing works.

robustini, 2019-12-21:
https://youtu.be/6mlW7paMfAA

3djc, 2019-12-21:
Yes confirm it is a issue while talking to the module in BL mode

robustini, 2019-12-21:
Hardware or software issue?

3djc, 2019-12-21:
Don’t know yet, someone with the right hardware is going to have to check that

gvt70, 2019-12-23:
Just updated to 2.3.4 and tried updating MPM with  
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/1.3.0.47

X12S: 
"Firmware update error"
NoSync

- same error with x10s on OpenTX 2.3.4
- same error with x9d+ on OpenTX 2.3.4

3djc, 2019-12-23:
I'm afraid that just means your module does not have a bootloader. I just flash mine from a x10 with 2.3.4.

robustini is also succeding from X10 btw

robustini, 2019-12-23:
> Just updated to 2.3.4 and tried updating MPM with
> https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/1.3.0.47/multi-stm-opentx-taer-inv-v1.3.0.47.bin
>
> X12S:
> "Firmware [...]

Thanks for your feedback.
Here 2.3.3 work fine on X10S Express and T16, no words.

robustini, 2019-12-23:
@gvt70 have you the right bootloader on your module?
Have you already flashed this module with your tx in a previous version of OpenTX?

gvt70, 2019-12-23:
> @gvt70 have you the right bootloader on your module?
> Have you already flashed this module with your tx in a previous version of OpenTX?

Yes, Flashed the same BIN when it came out a couple of weeks ago using X9D.
But, I have flashed it with a STK last week. Could that disable flashing with TX?

3djc, 2019-12-23:
You could have wiped the bootloader i guess

gvt70, 2019-12-23:
Sorry guys. BIG brain farth over here.
Just tested the MPM in a X9D again and it flashed fine.
X12S still no luck. Error as above. 
I will change about text to not cause any problems.

3djc, 2019-12-23:
Do you get Sync error or wrong signature ?

gvt70, 2019-12-23:
Just tested with X10S also. Same error as on X12S : Sync Error

3djc, 2019-12-23:
x10 works fine here

robustini, 2019-12-23:
X10S Express work fine here, so stranger.
But X12S no, so there's a issue... :-)

3djc, 2019-12-23:
Yes, but your X12 has a VERY different issue than NoSync

robustini, 2019-12-23:
> 
> 
> Yes, but your X12 has a VERY different issue than NoSync

No, see my video again, NoSync because i disabled the module in the second attempt... :P
Wrong Signature is the problem with the MPM enabled.

3djc, 2019-12-23:
Exactly what I'm saying, isn't it ?

robustini, 2019-12-23:
@gvt70, you have NoSync on X12S with the MPM enabled in the Module section?

gvt70, 2019-12-23:
I renamed the file to "multi-stm-v13047.bin" tried again. No luck. Tried a second time and it flashed succesfull with X10S. Tried flashing a couple of times and all went well.
X12S, same problem. First time it failed. Second time it flashed succesfull! Tried again and succesfull.

robustini, 2019-12-23:
> 
> 
> I renamed the file to "multi-stm-v13047.bin" tried again. No luck. Tried a second time and it flashed succesfull with X10S. Tried flashing a couple of times and all went well.
> X12S, same problem. First time it failed. Second time it flashed succesfull! Tried again and succesfull.

Ok, I try to insist on the X12S too, let's see what happens.

3djc, 2019-12-23:
The "Wrong Signature" message that you have means you have passed the 'noSync' stage, and that you are talking to the module bootloader, but the radio doesn't like/understand what it is receiving. NoSync and Wrong Signature are not at all similar

robustini, 2019-12-23:
No, it never works here, both MPM enabled and disabled, just different error, "Wrong Signature" when enabled and "NoSync" when disabled.
But I have ISRM on the X12S...

robustini, 2019-12-23:
Nothing, always "Wrong Signature".
I start to hate this signature check! :P

robustini, 2019-12-23:
> X12S, same problem. First time it failed. Second time it flashed succesfull! Tried again and succesfull.

Excellent, but he must always flash it, even on the first try.

gvt70, 2019-12-23:
Btw, the X12S and X10S still have the standard IJXT module. Not the upgrade module.
Maybe you could try with the standard X12S module.

robustini, 2019-12-23:
> 
> 
> Btw, the X12S and X10S still have the standard IJXT module. Not the upgrade module.
> Maybe you could try with the standard X12S module.

I've ISRM on X10S Express and work fine! :-)

robustini, 2020-01-03:
I tried today on a friend's X12S, on the first attempt with MPM enabled he gave me "No Sync", on the second attempt he wrote it.
There is therefore a problem introduced recently, with 2.3.2 he always flash it on the first try.

3djc, 2020-01-11:
@robustini what exact type of external multi are you using ?

robustini, 2020-01-11:
> @robustini what exact type of external multi are you using ?

I have tried with three different ones, it does the same thing, IRangeX IRX4, Jumper and URUAV TMX5.

markran, 2020-01-14:
I believe I'm having the same issue as @robustini. I have a IRangeX IRX4 and tried to flash the MPM firmware from OpenTX on my Horus x12s which has the original internal TX, not ISRM but got a "Firmware Update Error", "Wrong signature". I have bootloader 2.3.4 installed and "multi-stm-opentx-taer-inv-v1.3.0.47.bin" is on the MPM. 

I've tried doing it several times consecutively and also rebooting in between with no success. Any ideas?

robustini, 2020-01-14:
The solution would be simple, in OpenTX a warning of "signature wrong, flash it anyway?", with the possibility of skipping the signature check.
Or find out where the bug is...

3djc, 2020-01-14:
Again, no, signature check as nothing to do the firmware file, the signature is with the module, dialog with the module bootloader cannot be established, that cannot be bypassed

robustini, 2020-01-14:
Then the thing is even more critical.
Could it work randomly?
What was introduced in OpenTX to block communication?

3djc, 2020-01-14:
well you know if it work randonly more than us if it is random or not. From what you have been saying it looks like it nevers works so not random.

Nothing is introduced to block communication. To update, the module is started in bootloader mode, and your radio is than unable to talk to it, hence is (very not descriptive) error message

robustini, 2020-01-14:
If there is no possibility to log this event it will be difficult to understand where the problem lies.

markran, 2020-01-16:
@robustini Yes, I tried it both with and without the internal IJXT enabled and got the same error message.

@3djc said
> ... dialog with the module bootloader cannot be established

I'd like to help figure this out but I don't really understand how it's supposed to work. So a few questions if you don't mind.

1. In what ways is the dialog with the module bootloader different than the dialog with the module for normal control? For instance, OpenTX appears to communicate with the firmware on the module when it shows the firmware version number in the model screen. Maybe there's a clue in some difference in the handshake between the two instances? In both instances (bootloader and normal control) the X12S and MPM are communicating via the same electrical pathway. So then is the issue most likely a difference in the software on one side or the other or are there other places the problem could be? 

2. Do we know (or suspect) if the issue is with the particular X12S or is the issue with the particular MPM module? This MPM module was successfully flashed via FTDI so the flashing app was able to establish dialog with the module bootloader. Would it help if someone having this error can resolder the FTDI connections to the MPM and confirm that the FTDI can establish communication with the bootloader? @robustini Have you tried reflashing your MPM via FTDI?

3. I also have an original Taranis here as well as an X-Lite Pro. Would if help to try flashing this MPM from one of those (assuming I rig up some wiring to the X-Lite Pro)? Just trying to think of ways to narrow down the issue.

kilrah, 2020-01-16:
If you are sure your module has the bootloader version that allows flashing from the radio, and it works fully (control and telemetry/status) then it's not the module...

robustini, 2020-01-16:
> @robustini Have you tried reflashing your MPM via FTDI?

I write the same MPM on the T16 without any problems, so the bootloader is correct.

robustini, 2020-04-13:
No news about that?

robustini, 2020-04-13:
> @robustini Have you tried reflashing your MPM via FTDI?

I can flash this MPM with USB without issue, inserted in the slot of the X12S turned off!
It is not a problem of the MPM, I flash the same module from the tx menu in the X10S Express and T16, so the problem is with X12S and ISRM.
All friends who have the X12S with the original IXJT module manage to flash MPM from the tx.

kilrah, 2020-04-13:
Seems it always works if the external module is set to OFF in the selected model, just not if it's set to MULTI.

robustini, 2020-04-14:
> 
> 
> Seems it always works if the external module is set to OFF in the selected model, just not if it's set to MULTI.

It would be nice if it were so, unfortunately it doesn't work anyway.
OpenTX V2.3.8 (latest master of today), this MPM flashes smoothly with X10S Express and T16.
Same problem with the external T16 MPM module istalled on this X12S with ISRM.
The problem is this tx with the ISRM installed, not the MPM.

https://youtu.be/PQVdgLLxBKw

robustini, 2020-04-14:
> 
> 
> Seems it always works if the external module is set to OFF in the selected model, just not if it's set to MULTI.

It would be nice if it were so, unfortunately it doesn't work anyway.
OpenTX V2.3.8 DEV (latest master of today), this MPM flashes smoothly with X10S Express and T16.
Same problem with the external T16 MPM module istalled on this X12S with ISRM.
The LEDs of the MPM are clearly seen flashing in the video, so the TX tries to communicate but cannot.
The problem is this TX with the ISRM installed, not the MPM.

https://youtu.be/PQVdgLLxBKw

kilrah, 2020-04-14:
My X12S has ISRM installed and external multi flashing always works if the external module is OFF.

robustini, 2020-04-14:
@kilrah tranks for your feedback, for me it is law!
So do you think it could be that pin that I removed from the ISRM in order to use the multi trainer?

robustini, 2020-04-14:
@kilrah thanks for your feedback, for me it is law!
So do you think it could be that pin that I removed from the ISRM in order to use the multi trainer?

robustini, 2020-04-14:
Obviously this thing is relatively castrating, I prefer to have the multi trainer usable, so I will flash the MPM via USB in a moment.
It was only to understand if it is a problem caused by the absence of that pin, or if there is something else on my X12S.

kilrah, 2020-04-14:
Since it doesn't work when multi is on there is certainly something that's wrong. Kinda low priority though.

rmd4u, 2020-05-03:
Hello, I have the same problem with Horus X12S.

It is opentx 2.3.7 on Horus and I use the multimodul IRangeX IRX4.  The internal module is switched off and the external one is activated as Multi. 
After selecting flash file multi-stm-opentx-taer-inv-v1.3.0.76 or multi-stm-opentx-taer-inv-v1.3.0.91 the message "device reset" appears, then it makes a beep and the error message " wrong device" appears on the display. The same files with the same multimodule work on the T16 without problems.

[48 more replies]

closed 2021-03-20 by stale[bot]