opentx/opentx #7228: RX #0 shows as blank instead of 00 in Companion

Gfyss, 2019-12-21:
Issue with Companion 2.3.3

Slot 1 is assigned to receiver #0 but, as you can see in the screen capture, the RX# display is blank. Companion appears to be using that as the "unassigned" indicator.  Since receiver slot #0 is a valid setting and all new models are currently assigned that slot automatically, is there a better way to indicate a model slot is empty or not assigned a receiver slot, or pick the first RX slot not in use when creating a new model?

**Bottom line**: RX#0 is a valid selection and should be displayed instead of being blank.

[image]

In case it matter - Radio: X9E with JP4N1 module FW:1.3.0.47

kilrah, 2019-12-21:
The radio never chooses 0, and it's commonly considered as the "I don't want model match" choice.

Gfyss, 2019-12-21:
Ok, is it possible to indicate that in Companion?  At the moment, a blank entry can mean 1 of 3 things:

1 - receiver assigned to slot 0  (no model match/receiver lock guaranteed)
2 - no transmitter modules enabled (default after creating a new model)
3 - blank/empty model 

#3 is pretty obvious, but 1 & 2 can cause confusion, especially with the instructions to verify everything after upgrading.  I'm all for clarity, and assumptions tend to get muddled and shifted if not codified some how.  There should be a clear display difference between situation 1 and 2.  1 is flyable, 2 is not.

FYI: I just did the 2.2.4 -> 2.3.3 upgrade.  In the process of checking the models I noticed the new column in the .otx file, and that a few of my models had no RX associated.  I assumed something had gone wrong in the transfer.  I checked the radio settings and functionality, and everything does work so it's just a Companion display issue.  Now that the multi-protocol and Open-TX support more than 16 receivers I'll associate all the ones that are "sharing" RX#0 to their own slots. (Thank You all for doing that!!)

closed 2020-10-12 by stale[bot]