opentx/opentx #7230: Companion 2.1 does not download FW

muelli01, 2019-12-22:
When trying to download firmware
(e.g. opentx-9x-frsky-nofp-nocurves-battgraph-pgbar-novario-en-2.1.9.hex) the companion just shows
"Unknown server failure, try later".
A file is however created with content "ERROR:125".

Tried bothe the win32 and Linux64 versions.

3djc, 2019-12-22:
Could you try again please ?

muelli01, 2019-12-22:
Still the same error....

3djc, 2019-12-22:
Yes, I need to find a source for those version old compilers

muelli01, 2019-12-22:
I am not quite sure which version is installed right now, I selected the build options as good as i remember......Is there a way to find the exact version?
It could as well be "opentx-9x-frsky-nofp-nocurves-rotenc-battgraph-pgbar-novario-en-2.1.9.hex" which I need.... Sadly I cannot find any version info in the remote itself.

Thanks!

kilrah, 2019-12-22:
Nothing to do with you, our servers just does not have what's needed to build AVR stuff anymore, need to figure out how to restore that capabiltity from files that are so old the known source doesn't exist anymore.

muelli01, 2019-12-22:
OK, understood..... thanks for trying to figure this old stuff out :)

I just discovered the old hex file I was using (opentx-9x-frsky-heli-nofp-nocurves-ppmca-ppmus-symlimits-autoswitch-battgraph-pgbar-novario-nogps-en-r2940.hex) so options are clear. If upgrade to 2.1 is not possible because "old stuff gone away", then it is just bad luck :)

3djc, 2019-12-22:
Could you give it another try ?

muelli01, 2019-12-22:
It downloaded sth, will try to flash asap

muelli01, 2019-12-22:
OK, had a little trouble flashing, but after fixating to 32kHz i could flash the FW and it is working.
Thanks alot, appreciate your help!

3djc, 2019-12-22:
Great ! An our appologies for the lost time, have great flying !

closed 2019-12-22 by 3djc