opentx/opentx #7234: Bootloader not accessible with new X9D Plus SE - 2019

Cattlebaron, 2019-12-23:
With the new on/off switch on the TX one can not enter the bootloader mode to update the the Firmware. At present the TX is running on OpenTX Version 2.3.0

3djc, 2019-12-23:
We have many report that it is working fine. Maybe you can download a 2.3.3 firmware, put it on the sdcard, navigate to this file from OpenTX click on it and select flash bootloader. That should  rewrite it in case your is defective for some weird reason

Cattlebaron, 2019-12-24:
Thank you for the information and advice but alas it did not work on my TX with Bootloader mode. I also followed the OpenTX (I assume OpenTX Companion) procedure suggested and get a file write error.
[image]

MRC3742, 2019-12-24:
Seems from your picture you have the wrong radio selected in the "Edit Settings"
You have Radio: Taranis X9D+ and it should be set to Taranis X9D+ 2019

If you downloaded firmware under this Radio Type then it is not correct for your 2019 Radio.
Change and Redownload firmware and then try to update.

If that doesn't work for you then.....
You can just update the bootloader from within OpenTX on the Radio as 3djc described above using the new firmware file.

Rich

driedeker, 2019-12-25:
both trims and press the on off twice in sucsesion seam to work for me

Cattlebaron, 2019-12-25:
Thanks a million times for the advice. I will work through the issue over the next day or two, I had reached a blank wall as far as I could see. I hope it will help others who may have experienced or are yet to experience the issue.

Cattlebaron, 2019-12-26:
To add to the above post, thanks to driedeker I had the best Christmas present this year. I works like a charm, how did you find that method out?  I posted requests on RCGroups, intoFPV and openrcfotums without any useful replies before I asked here so thanks again.

Hank

kilrah, 2019-12-26:
Still is an issue since it's ot how it's supposed to be

driedeker, 2019-12-26:
> 
> 
> To add to the above post, thanks to driedeker I had the best Christmas present this year. I works like a charm, how did you find that method out? I posted requests on RCGroups, intoFPV and openrcfotums without any useful replies before I asked here so thanks again.
> 
> Hank

I was lucky as I found it by trying all else.
Yes its not how it should work,but a work around.

Cattlebaron, 2019-12-27:
Just officially document the "work around" and it will then be the default method of access to the bootloader.

closed 2020-10-12 by stale[bot]