opentx/opentx #7239: There is no sub-protocol for FRSKY 2A in version 2.3.4 for the internal module.

PhATL1, 2019-12-24:
There is no sub-protocol for FRSKY 2A in version 2.3.4 for the internal module.
[image]
[image]

3djc, 2019-12-24:
Could you do the recommended module update?

PhATL1, 2019-12-24:
I tried using the Jumper firmware from their site, but got no change and then I realized that I may have made a mistake and should have used an update from you guys instead. I'm just totally confused
Eddee HorseyOwner, DesignerHorsePower Studios817.525.5875

3djc, 2019-12-24:
I'm talking of the multimodule upgrade, as asked by the radio on your screenshot

3djc, 2019-12-24:
You need to get a multi module firmware from https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases and flash it on your module. How you do that depends of the actual version of  your module

PhATL1, 2019-12-24:
thanks I'll try that and let you know. You guys freaking rock

3djc, 2019-12-24:
I'm closing this as it is not an OpenTX bug by the look of it. You can find help here : https://www.rcgroups.com/forums/showthread.php?3243685-Jumper-T16-2-4G-16CH-OpenTX-Multi-protocol-Radio

closed 2019-12-24 by 3djc