opentx/opentx #7240: Select Input missing in SF Adjust GV

ChristiaanVis, 2019-12-24:
Minor issue I think:

Running OTX 2.3.4 on a FrSky Horus 10S (with ISRM module) it seems not to be possible to select inputs or controls from the list as a source to adjust a GV in special functions. This is possible in Companion.

Companion:
[image]

Horus X10 running OpenTX 2.3.4
[image]

Also noticed the '???' in Companion listed as controls. On the tx it seems that these entries are listed as not used pots named EX1 and EX2?
[image]

kilrah, 2019-12-24:
You're already in Source mode, the popup is to switch to another....

closed 2019-12-24 by kilrah

ChristiaanVis, 2019-12-24:
Ai, you are right of course. Sorry for the spam.