opentx/opentx #7242: TR5 and TR6 can't be use to update a GV

LapinFou, 2019-12-28:
Hi folks,

I'm reporting a bug from the French forum.
The history is here: http://frskytaranis.forumactif.org/t9699-ajustage-fonction-globale

If you program this:
SF1 | ON - Adjust GV1 (Trm5)
The GV1 is not update. It does work for TrmA, TrmT, etc... bu doesn't for Trm5/Trm6

This bug can be reproduce on the simulator and on a Horus X12S.

kilrah, 2019-12-28:
Working fine on radio. 
On simulator the buttons for T5/T6 do not work , known issue. https://github.com/opentx/opentx/issues/7167

closed 2019-12-28 by kilrah

afdofdez, 2019-12-28:
Horus X10S Express with OpenTx 2.3.4 GV adjustment works perfectly with the TR5 and TR6.
In the simulator it does not work.

LapinFou, 2019-12-28:
Thanks guys. I will cross-check with the French member (I don’t personnaly own a Horus)

LapinFou, 2019-12-30:
The French member has been confused between **TR5** and **Trm5**.
So, there is no bug on the radio firmware. There is only a bug in Companion.