opentx/opentx #7244: X9 Lite S companion warning about documentX.otx from radio

johnelliott, 2019-12-29:
**Describe the bug**
Companion gives a warning that an x9 lite s radio is an x9 lite. This seems erroneous. Found after testing results of #7215.

**To Reproduce**
Steps to reproduce the behavior:
1. Reset X9LITES to factory defaults with OpenTX 2.3.4
2. Open companion of 2.3.4 release
3. Click read models and settings from radio
4. See error dialog box

**Screenshots**
[image]

**Expected behavior**
No error dialog box

If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.3.4 from releases page
 - Hardware Frsky X9LITES
 - OpenTX Companion version 2.3.4 from releases page

kilrah, 2019-12-29:
[image]

They are considered as different radios, configure for the right one.

closed 2020-01-06 by bsongis

johnelliott, 2020-01-06:
Thanks Bertrand. I had trouble getting my builds set up but perhaps I can help out with things like this in the future.

MrMagniv, 2020-12-03:
Just got a new x9-lite-s and had the same problem, is there a way to fix this?

kilrah, 2020-12-03:
Update both companion and firmware to latest.

MrMagniv, 2020-12-03:
I'm using companion 2.3.10, is this good?
also, how do I check what firmware I have?

MrMagniv, 2020-12-03:
the fw I have is opentx-x9lites
and the vers is FrSky (d39cc6fa)

kilrah, 2020-12-03:
The frsky stock build probably doesn't have the fixes done here so you should update with our 2.3.10.

commit ae8fa86a (2020-01-06): Attempt to fix #7244
commit 71c763f9 (2020-01-06): Bsongis/x9lites sources companion fix (#7275)

Fixes #7215
Fixes #7244