opentx/opentx #7247: Trim 5 Trim 6 issues across Flight Modes

Stuntman2, 2019-12-29:
Companion 2.2.4 Horus x12s
Value of Trim 5 is not remembered in FM1 thro FM8
Value of Trim 6 is not remembered in FM1 thro FM8
When you have transferred the model to the radio you can go into Flight Modes and manually change the trim setting for Trims 5 & 6 and the transmitter will then function properly until the next time you write Models and Settings to Radio.

kilrah, 2019-12-29:
Which version? 2.2.4 isn't supported anymore.

Stuntman2, 2019-12-29:
I have upgraded to 2.3.4 and the issue is still there.

Stuntman2, 2019-12-29:
I have upgraded to 2.3.4 and the issue remains

3djc, 2019-12-31:
Could you give more details ?  I have programmed this :

[image]

Using 2.3.4, if I read radio and settings, and write back, this is still unchanged. Am I misunderstanding your issue ?

afdofdez, 2020-01-02:
Horus X10S with OpenTx 2.3.4. 
When writing a model on the radio or turning on the radio, I don't know at what time, change the existing value of GV9 in FM0 to the value 1. 
In the two models that happened to me, GV9 could be modified by T6.

Stuntman2, 2020-01-03:
[GitHub.zip](https://github.com/opentx/opentx/files/4018831/GitHub.zip)

Stuntman2, 2020-01-03:
Sorry for the late reply @3djc @afdofdez, I have attached both an otx and a word document which I hope will explain the problem in more detail.

3djc, 2020-01-04:
Simulator does not store anything, so if you change value there, they will not be changed when sent to radio

closed 2020-01-04 by 3djc