opentx/opentx #7249: taranis 2019 se has no option to use LR12 protocal internal module.        2.3.4

williajgr, 2019-12-29:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

kilrah, 2019-12-29:
Because the module doesn't support LR12.

closed 2019-12-29 by kilrah

3djc, 2019-12-30:
FrSky has decided that the ACCESS modules will  not support LR12, not an OpenTX decision

williajgr, 2019-12-30:
Frsky's taranis 2019 firmware supports lr12 protocal and I think 2.3.0 aupports it as well.

3djc, 2019-12-30:
We were initially told lr12 was going to be available with access modules so we added the option for 2.3.0. Then we where later, but too late, informed it was not the case. So yes there is an option in 2.3.0, but no it never worked because access modules never supported it

williajgr, 2019-12-30:
That is a big miss for frsky.  I think they are a little out of control.  Their support team never answers the phone.   thanks for the update.