opentx/opentx #7250: when converting models from tranis to tranis 2019 internal transmitter is turned off and all sensors are set to external.

williajgr, 2019-12-29:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

kilrah, 2019-12-29:
As designed.

williajgr, 2019-12-30:
Why would one design it that way?  I have 30plus models each with at least 5 sensors.  This requires I go in and edit more than 150 sensors.  This is a lot of work.  Actually it is worse than that as the all of the id’s have to be changed from 2xx to xx.
    Thanks for the input.
        Joe

kilrah, 2019-12-30:
Because the ACCESS system is different and sensors need to be assigned to a specific receiver, which can not be done until one is bound. 
You should not touch the IDs.

williajgr, 2019-12-30:
Another problem with what you are proposing is I will have to redo all of the telemetry pages and the logical warning pages.  Much easier to change the id’s as long as I am not using access protocol.  I have done this on several models and it works perfectly, but still a lot of work. 
  Thanks, and please consider fixing it.
      Joe

williajgr, 2019-12-31:
Any chance you will fix this??

3djc, 2019-12-31:
We cannot change how access works, so probably not

williajgr, 2019-12-31:
I am not using access. Is there any chance we could discuss this over the phone?    9143887776

williajgr, 2020-01-02:
Doesn't seem like you understand what needs to be changed to solve the problem.  When you do the conversion why can't you go from d16 to d16 instead of d16 to off? and have the sensors use the internal transmitter?  I can edit the sensors to the correct id's and i don't have to rediscover any of them. This works perfectly. seems like this would be easy to do.

bsongis, 2020-01-02:
When only internal module used:
1) Internal module "XJT / D16" could be converted to "ACCST / D16"
2) Telemetry sensors ID could be attached to moduleIndex=0 / rxIndex = 0
@3djc am I wrong?

bsongis, 2020-01-02:
The BIND will still be needed

kilrah, 2020-01-02:
Seems OK

3djc, 2020-01-02:
I would leave the module off, this way you can track those who have been rebinded. Sounds good for sensors

bsongis, 2020-01-02:
yes agree with last comment from 3djc

williajgr, 2020-01-02:
sounds all good to me.  It will be good to set it to accst d/16 rather than off, then I won't have to set the fail safe again.  Just bind and be done.   I do wish there was a way around binding again, as I have 30+ models that I have to dig into to get to the bind pin on the receiver. I guess you don't have access to the bind codes in companion but that would be sweet.  On some of my old Taranis's I just swapped the transmitter boards and skipped the rebinding process. This works well.
   Thanks again for the help

bsongis, 2020-01-02:
I don't think we loose the failsafe values when setting to OFF (failsafe values don't depend on the module type)

williajgr, 2020-01-02:
No the fail safe codes are not lost, but the fail safe is set to "not set" so you have to go in and edit it back to what you had set it to before.  I don't always set it to the same thing so I have to bring up the fail safe settings on my old radio to set it the same.  Some times I use custom sometimes I use hold and sometimes I use no pulses. This information is lost.

bsongis, 2020-01-03:
true

kilrah, 2020-01-03:
> I guess you don't have access to the bind codes in companion but that would be sweet.

The bind codes are built into the modules and are not accessible in any way.

williajgr, 2020-01-03:
Maybe one could get frsky to make the bind codes accesable in some future hardware.

bsongis, 2020-01-03:
we already have this code for internal radio conversions:
```cpp
#if defined(RADIO_X9DP2019)
      if (i == INTERNAL_MODULE) {
        newModel.moduleData[i].type = MODULE_TYPE_ISRM_PXX2;
        newModel.moduleData[i].subType = MODULE_SUBTYPE_ISRM_PXX2_ACCST_D16;
      }
#endif
```

kilrah, 2020-01-03:
> Maybe one could get frsky to make the bind codes accesable in some future hardware.

This would be a major safety issue, the whole point of bind codes is for them to be unique so as not to risk having 2 radios with the same on the same field. If people can change it anything can happen.

The new ACCESS system allows for easy use of 2 radios without that risk.

williajgr, 2020-01-03:
You are probably right.

closed 2020-10-12 by stale[bot]