opentx/opentx #7251: No support for internal gyro on horus x12s

williajgr, 2019-12-29:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

williajgr, 2019-12-31:
Nobody responds to Horus problems??

1Sven, 2020-01-03:
I am also very interested in a solution.

strgaltdel, 2020-05-14:
Hi,

would be very interested in X12 gyro implementation too,
guess it could simplify the compatibility of models between X-LiteS, whose Gyro is already supported.

is an implementation still planned ?
No milestone any more !

thanks & regards

1Sven, 2020-05-14:
It is possibele!!! Raphael Coffig has it already done for me on one raining day in the hollydays... But only in a old Version and without input. I have a luascript for that. I hope and be still waiting that it will comes in the future... For me its the "Bierschalter"... You can fligt and do anything at the same time...

strgaltdel, 2020-05-19:
Hi 1sven,

interesting,
does it mean, "your" solution is based only by expand lua functionality (or using undocumented lua features),
or did Raphael some coding in oTx ?

regards

notlebowski, 2020-05-19:
I was interested in using the gyro to detect a DLG launch, but since I bought an X-Lite I never use that heavy radio anymore for that purpose. Still... I was a bit disappointed when I noticed no support would be given to it.
Before buying a radio you check the features and you think all of it will get support eventually. (_eventually is not equal to éventuellement_)

1Sven, 2020-05-19:
Hello,

 

Rafael has made a special software for me. To use the function is it necessary to have a little Lua-script.

It’s a funni thing.. but the developers at OTX are not interesting and the basic- SW was about 2.0 or so.

For the time I’m using 3.7.

 

Sven

1Sven, 2020-11-18:
Please weake it up for the future... may be 2.4?

jbule1980, 2021-03-28:
> _Before submitting, search the open issue if there is already a report of the bug_
>
> Also often minor bugs are already fixed in nightly builds, often it is
> a good idea to test (in a controlled [...]

Put back accelerometer and Gyro on openTx! It's very usefull for no headtracking FPV googles!

strgaltdel, 2021-03-29:
Ich kehre zurück am 06.04.2021.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] No support for
internal gyro on horus x12s (#7251)" gesendet am 29.03.2021 01:47:15.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

islandflight, 2021-06-05:
Please would you add gyro function to a logical switches go enable telemetry call outs etc on tilting the x12s as is the case with the x20.

raphaelcoeffic, 2021-06-05:
> interesting, does it mean, "your" solution is based only by expand lua functionality (or using undocumented lua features),
or did Raphael some coding in oTx ?

There was some coding in OTX involved ;-) Currently, my plan is to add this feature to [EdgeTX](github.com/EdgeTX/edgetx).

Luznatural, 2021-09-06:
I continue interested on this feature...

strgaltdel, 2022-04-16:
Ich kehre zurück am 21.04.2022.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] No support for
internal gyro on horus x12s (#7251)" gesendet am 16.04.2022 14:53:25.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

Luznatural, 2022-04-18:
Please, do not close it.
Thanks in advance