opentx/opentx #7252: Telemetry Screen Switching not working. (Works Once)

LawrenceLDJr, 2019-12-30:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
Inability to cycle  through the 4 telemetry screens on the Jumper T12. Holding down a few seconds brings up telemetry screen 1, up or down will change the screen once then you stuck in that screen unable to cycle. You can exit to home screen and everything else is fine. Entering telemetry with down button, the screen that locked will be there and is working. But you can't cycle at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Show Telemetry...'
2. Click on 'Down Button....'
3. Scroll down to 'Thats it ....'
4. See error 

**Expected behavior**
To cycle through the 4 possible telemetry screens.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.3.4
 - Hardware Jumper T12
 - OpenTX Companion 2.3.4

3djc, 2019-12-30:
Is one of your screen a non telemetry script ? (ie a script like BF that needs key inputs ?)

PasiRuo, 2019-12-30:
I can reproduce the behavior with inav telemetry script

3djc, 2019-12-30:
Yes those are not telemetry scripts as they do use inputs

PasiRuo, 2019-12-30:
script that works and does not reproduce the behavior?

3djc, 2019-12-30:
Telemetry scripts, any that does not have input and only displays telemetry items in a nicer way than raw data

PasiRuo, 2019-12-30:
Scripts run but create situation LawrenceLDJr reported:  You're stuck in that screen unable to cycle.

including "Telemetry scripts, any that does not have input and only displays telemetry items in a nicer way than raw data"

LawrenceLDJr, 2019-12-30:
> Is one of your screen a non telemetry script ? (ie a script like BF that needs key inputs ?)

Yes I have available Betaflight, INav, Farl scripts... which did cycle before I updated. Always ran 2 info screens with INav and Farl. Now I see the with 2.3.4 keypad changes things are different. I can run more than 1 info screen, since the do not have key inputs. THX 2.3.4 key inputs for the T12 are nice, I'm glad the speed key is back. Holding enter on the entry fields doesn't bring up as many options. Lua scripts with key commands seems cannot co exist with the new input config. Maybe if enough OTX junkies ask for an update these issues will be fixed. THX again

LawrenceLDJr, 2019-12-31:
INav and Betaflight Lua Telemetry screens still work. Betaflight totally functional but locks, INav functions but only enter key works to cycle through INav screens, cannot bring up INav setup screen even if you execute INav Lua, selecting Max and current data is gone too. Farl telemetry screen works but locks when selected, strange since this screen doesn't have key inputs.

PasiRuo, 2019-12-31:
> Farl telemetry screen works but locks when selected, strange since this screen doesn't have key inputs.

LawrenceLDJr, please reconsider and open this issue as it is unresolved. 

Simple test to reproduce. I believe it should allow cycle past this screen, but does not: 

`
local function run(event)
  lcd.clear()
  lcd.drawText(1, 1,"Text1", 0)
end

return{run=run}
`

LawrenceLDJr, 2019-12-31:
Did not close on purpose but saw it did close and didn't know how to re open. INav issue I mentioned above is an INav issue. INav Devs will make the changes needed to match OpenTXs new key set. Maybe I'm spoiled but I want those screens 1 thru 4 functional again. As I stand corrected if you only use the functions available within OpenTX you can cycle through all 4 screens. So far the use of Farl, Betaflight or INav your screen will not cycle once one of these are selected.

PR #7263 (2020-01-06): Fix some keys no sent to telemetry screens handler
This fixes #7252
closed 2020-01-06 by bsongis