opentx/opentx #7253: TX/RX data from Taranis-receiver

villamany, 2019-12-30:
Hi, is there any way for sending data from the Taranis serial port to receiver?
I got a bluetooth module and inverter installed on my x9D+ and im receiving the SPORT telemetry from my iNav plane into my phone, but i would like also to send data to the plane from the phone. I don`t now if this is possible by SPORT or MSP yet but i dind´t found any extended info about taranis UART (battery slot pins). I see only "TELEM.MIRROR", "TELEM IN", "SBUS TRAINER", and "LUA" on the Serial port screen configuration but not sure if any of these options works like bidirectional port.

Thanks

3djc, 2019-12-31:
Well there are many steps needed for something like that to work. Currently there is no way to enter data into BT, you would then need to way  to encode so it gets transmited using telemetry, then you will need somekind of decoding protocol on the receiver end. Also you need to be aware that throughtput would be extremely extremely low

villamany, 2019-12-31:
is no way to do this though bidirectional SPORT?

kilrah, 2019-12-31:
yes but then it would be independent from OpenTX.

brainbubblersbest, 2020-05-04:
This will be in need anyway, if access functions are full Integrated. 
[image]

I expect Serial telemetry link like the one in crossfire. 
But crossfire uses it's Own BT Module for that. 
Anyway. With HM-10 Module paired with internal x12s BTLE, crossfire lite AND Telemetry Mirror. This is working too.

closed 2021-01-30 by stale[bot]