opentx/opentx #7255: Add an Increment Timer function in addition to  Set/Get Timer

PaoloFCantoni, 2020-01-02:
If using timers while flying (rather than telemetry - battery usage etc.) you may need various timers for the various batteries you use.  Sometimes, you might forget and use the wrong one (ask me how I know :()  It would be useful to be able to adjust the current value of the timer up or down.

**model.incrementTimer(timer, value)**
Increment/decrement model timer parameters

Parameters
timer (number) timer index (0 for Timer 1)
value timer parameters, see model.getTimer()
*NOTE: If the value is positive, increments the timer, if the value is negative, decrements the timer.*

Return value
none
Notice
If a parameter is missing from the value, then that parameter remains unchanged.

xoedusk, 2020-01-07:
I'd also like to see this. Would be nice to have it implemented as a built-in function available in Special Functions.

khalllled, 2020-06-25:
hi

closed 2021-03-26 by stale[bot]