opentx/opentx #7256: Companion 2.3.3 2.3.4 X7access firmware download error

mStrangers, 2020-01-02:
Hi 
i can't download firmware for x7access ( every compilation option) with companion 2.3.3 or 2.3.4.
i had always a invalid board error and error:6 when i use this URl directly : https://downloads.open-tx.org/2.3/release/firmware/getfw.php?fw=opentx-x7access-en.bin

3djc, 2020-01-02:
Looks like FrSky forgot to add that to compile options

mStrangers, 2020-01-03:
This didn't work completely , i can compile the firmware , upload it to the radio , it work on it.
but there's no access or accst protocol selection for the internal module only XJT , i've check the compilation option and didn't find where is the one for enabling access protocol.

kilrah, 2020-01-03:
This fix is only for the server, the code was always OK.

mStrangers, 2020-01-03:
Ok so can you give me the Gmake option for activating ACCESS , when i'm build it there no ACCESS protocol for the internal module.
Compilation option USED : -DPCB=X7 -DPCBREV=ACCESS

closed 2020-01-03 by bsongis

kilrah, 2020-01-03:
Should work like that, make sure you cleared your build directory before launching cmake

3djc, 2020-01-03:
[image]

We do not have any x7 access radio, but it should work

mStrangers, 2020-01-03:
hi , re clone the git , and this work thanks a lot guys

PR #7257 (2020-01-03): Fix x7access remote compile
This fixes #7256