opentx/opentx #7260: Timer - vibrate every minute over (countdown) and more

Eldenroot, 2020-01-02:
I used Spektrum for a decade, it was a good radio but I switched to Frsky... one of the feature what I missing is better vibrates/sound notifications for timers.

Spektrum had an option on read the timer every minute, vibrate every minute over the set time etc. Is possible to get something like this (I can get a list of options from Spektrum DX9), or am I missing something in the OpenTX and this is already possible?

Thank you!

kilrah, 2020-01-02:
[image]

closed 2020-01-02 by kilrah

Eldenroot, 2020-01-03:
@kilrah - OK, I understand... anyway in Spektrum I can set vibrate+voices together, is that possible in OpenTX?

kilrah, 2020-01-03:
When on voice it will vibrate as well, try things...

Eldenroot, 2020-01-03:
Then it should be Vibe/Voice not just Voice :)

Thank you anyway :)