opentx/opentx #7265: Allow faster access to a few key scripts

3djc, 2020-01-04:
TOOLS is a step in the right direction, by providing simpler access to a large set of scripts, but people often have 1 or 2 key scripts that they use a lot for a specific model, and it would be nice to have a faster way to access those

3djc, 2020-01-04:
Maybe add a way to add 1 or 2 scripts in the long enter menu ?

lshems, 2020-01-04:
Every onetime script can also be run as a telementry script.

Isn't this the same requirement as asking for the telemetry screens to be accessable by the page button and have the radio store the last used screen as startup screen?

just add your so important script to a telemetry screen.

3djc, 2020-01-04:
Or not, we are going to restrict keys allowed for telemetry scripts, so unless they are purposefully taking that into account, they might not work

lshems, 2020-01-06:
Could you clarify on this restriction that is foreseen?

3djc, 2020-01-06:
Navigation keys (page, page down’ mdl long, sys long, tele long, ...) will be excluded. That’s the overall principle, details are yet to be finalised

lshems, 2020-01-06:
As long as the plus/minus, with long, release and first options and enter/exit with long, release and first options and equivalents, are supported, most user scenarios are supported. (of course the equivalent to plus/min etc depending on radio type)

Currently I use a long enter to open a text file to give help on that particular item to edit or menu item. It was always annoying to get the telemetry reset options pop up. An option is to not have the long press enabled for telemetry reset in a script page, only on the real telemetry value pages. The script can manage itself if a reset of something needs to be triggered on a long press.

So it would be appreciated to consider long enter to remain available as well, ideally with suppression of the reset telemetry popup.

there are 4 pages available, so using the normal navigation and having them in the main navigation sequence allows people to create apps with up to 4 pages running as separate apps. They share the same LUA space, so no major problem there.

if someone wants to navigate on app-specific pages, they then need to create a menu structure to select from, or use an previous/next item on your page.

suggestions and opinions, as usual. No request. Do as you wish.

klutvott123, 2020-01-18:
@3djc How about making TOOLS the first page? That way we could have faster access to all scripts. I find that TOOLS is the page I use most frequently. The other pages are mostly just set and forget(for me at least).

3djc, 2020-01-18:
I like the idea, lets see what the others have to say

kilrah, 2020-01-18:
Works for me

lshems, 2020-01-18:
Important: TOOLS are run as onetime scripts. They exclude mixer scripts.
Telemetry script don't exclude mixer scripts.

People might start flying while running one time scripts, and having problems with their mixer scripts and special function scripts to be executed.

Why not make the Telemetry screen part of the main pages sequence? 
Why not push to telemetry scripts instead of one-time? They are much more versatile?
Why keep the difference?

klutvott123, 2020-01-19:
One-time scripts are good for scripts that doesn't have to run all the time. That persistent scripts are stopped when running a one-time script is very useful and allows us to make good use of the available memory.

lshems, 2020-01-19:
That's why you should not have them on the main screen. One time is one time.

Telemetry is for all the time.

klutvott123, 2020-01-19:
Sure. I didn't say anything about putting them on the main screen.

lshems, 2020-01-19:
No you didn't, but this proposal is, or I misunderstand

kilrah, 2020-01-19:
No it's not, it's just about having a shortcut to launch a one time script more easily.

lshems, 2020-01-21:
By making it the main screen? Or first screen? Or is that the same?

klutvott123, 2020-01-21:
By making TOOLS the first page of the radio menu. 
So that long press menu takes you directly to the TOOLS page.

lshems, 2020-01-21:
Ok. Confused by wording. First, main, ..

dukey32123, 2020-01-22:
I'm already running and testing PR #7312 which makes it only two quick clicks to access Crossfire menu in my case, way more convenient!

kilrah, 2020-01-22:
> Ok. Confused by wording. First, main, ..

Nobody talked of "main" apart from you...

closed 2020-02-10 by 3djc