opentx/opentx #7266: Antenna defective error when internal RF ON

slixk, 2020-01-04:
After installing, or updating to version 2.3.3 or 2.3.4 on my EU Version Horus X12s, i get a visual and auditory error telling me, that the antenna would be defective (spoken error is "Radio Antenna defective"). This only happens, when I turn on the internal RF-Module. BUT I've never tried binding or connecting a RX on the 2.3.x, since I get the error...

When testing with Version 2.2.x, it works just fine, and I have the normal range, that I'm expecting.

Steps to reproduce the behavior:
1. Installing Version 2.3.x on the X12s
2. Start Radio
3. Turn on internal RF
4. See error every 5 (or so) seconds

**Expected behavior**
No error when I turn on the RF.

**General information**
 - OpenTX version 2.3.3, 2.3.4
 - EU Version FrSky Horus X12s

3djc, 2020-01-04:
Have you set antenna to be external in RADIO/HARDWARE/Antenna ? This is the first time we are hearing that, so very likely an issue with your setup

slixk, 2020-01-04:
@3djc You mean in the transmitter itself? I dont want to use an external antenna. I'm using the internal on 2.2.x and there's no problem whatsoever.

3djc, 2020-01-04:
yes, on the transmiter itself. I understand you don't want to use it, but it looks like it could be set to use it. I can't think of anything else that would create that error

slixk, 2020-01-04:
Oh I see, i'll have a look at it. thanks 👍🏼

slixk, 2020-01-04:
@3djc Strange... I cannot set the antenna (internal/external). Not in Model/Model Setup/Internal RF nor in RADIO/HARDWARE/Antenna (doesnt even exist). Also, I can no longer choose D8 mode, only D16 and ACCESS, which isnt even available for the X12s?

3djc, 2020-01-04:
Wait, you have flashed with option internalaccess with is very worng unless you have replace the internal RF with the upgrade kit

slixk, 2020-01-04:
Ohhh yes, thats it. I may have set this in the previous versions and it never was a problem...

3djc, 2020-01-04:
DO NOT DO THAT !

closed 2020-01-04 by 3djc