opentx/opentx #7268: BattCheck Widget - "Lowest Data" does not reset when 'reset cels' or 'reset telemetry' is activated

jkhoward2, 2020-01-05:
Appears to be a bug in 2.3.4 or 2.3.3.  Was working fine in 2.3.2  "Lowest data" in the BattCheck widget either shows zeros or lowest value (sometimes one, sometimes the other - can't figure out what triggers either), but 'Lowest data' values are not reset by activating custom function 'reset Cels' or 'reset Telemetry' as they used to be reset.

3djc, 2020-01-05:
To be sure we understand correctly, does it affect telemetry item (in telemetry screen) or what is displayed by BattCheck ?

jkhoward2, 2020-01-05:
It affects what is shown in BattCheck. Thank you for looking into this.

3djc, 2020-01-05:
@offer-shmuely could you have a look ?

offer-shmuely, 2020-01-05:
This is correct  :-(
I don’t remember that the old widget succeeded to reset, but the current one does not know the “reset event”

We open a request to get this event 
https://github.com/opentx/opentx/issues/6415

I will be happy to implement a temporary workaround if we can find a good one
Any suggestions?

3djc, 2020-01-05:
if your recorded minimum is bellow the - sensor, then reset did happen, or something along those lines ?

jkhoward2, 2020-01-05:
Sometimes, it just showed zero for all six cells (I connect the receiver to
power before connecting the balance lead to the MLVSS sensor). However,
other times, the reading was the lowest cell voltage.  In both cases, the
'lowest data' would not reset upon reset activation.  The only way to reset
the readings was to disconnect all power....but upon reconnecting power, it
was often zero volts for each cell.

Fyi, hardware is X10S, S8R (current firmware version), and MLVSS. Battery
used in plane is 6S 5000mah.

3djc, 2020-01-05:
Well the sensor should be connected first isn’t it ?

jkhoward2, 2020-01-05:
Maybe. I'll try that shortly and let you know. That being said, ive been
connecting in this order for months and never had a problem resetting the
lowest data. I would think the reset would occur upon demand regardless of
the connection order.

I'll get back to you asap after I try again. Thank you.

jkhoward2, 2020-01-05:
Just now, I tried connecting the MLVSS to the battery balance lead before
connecting the main battery lead to the ESC and BEC. I also tried
connecting a spare MLVSS sensor I have to see if that made any difference.
I also tried connecting the main battery lead to the ESC and BEC before
connecting the battery balance lead to the MLVSS (just as I had been doing
in the past).

No matter the order of connection or which MLVSS I connected to, the
'lowest data' reading was zero for each cell and it did not change by
resetting Cels or resetting Telemetry.

3djc, 2020-01-05:
I guess for this specific widget, 0 should not be considered a valid low cell value maybe

jkhoward2, 2020-01-05:
Yesterday, even when actual cell values were listed for 'lowest value',
they would not reset on demand.

offer-shmuely, 2020-01-05:
jkhoward2, as I wrote, you are correct, it’s a bug, no need for more testing.
I will find a way to fix it.

jkhoward2, 2020-01-05:
Thank you very much.

offer-shmuely, 2020-01-26:
The old widget was reseting the lowest value each time the telemetry stopped (turning off the rx battery), so it did not give a chance to investigate the battery after landing.
The new widget is designed to enable this behaviour, but that in a cost of needing to implement a reset (an API that still not exist)
I am experimenting with a method that will enable to reset it.

jkhoward2, will you be able to check a fixed version?

jkhoward2, 2020-01-26:
Absolutely.  Thx!

offer-shmuely, 2020-01-26:
https://github.com/offer-shmuely/opentx/tree/shmuely/horus_batt_widget_reset_telemetry/radio/sdcard/horus/WIDGETS/BattCheck

jkhoward2, 2020-01-27:
Maybe I am not using the script appropriately, but it is not working on the “lowest cells”.  However, the total pack voltage is resetting (I was not watching that value before, so I don’t know if that is a change).  See attached photos:

 

*	‘Lowest data_by cell.jpg’ – this does not reset upon activation of the script (just like before)
*	‘Total pack voltage.jpg’ – this does reset (although it may have worked before – I did not watch this one)

 

Process I used with script you sent:

 

1.	Clicked on your link
2.	Clicked on the file name
3.	Clicked on “Raw”
4.	Saved as “BCkRes.lua”
5.	Copied file to SDCARD\SCRIPTS\FUNCTIONS
6.	In Special Functions on X10S, set a switch to Lua Script
7.	Designated “BCkRes.lua” as the script

 

https://github.com/offer-shmuely/opentx/tree/shmuely/horus_batt_widget_reset_telemetry/radio/sdcard/horus/WIDGETS/BattCheck

offer-shmuely, 2020-01-27:
1. Take the script again, i added a fix and not sure if you took it already
2. After downloading the scrip, you need to put it under the WIDGET/BattCheck/main.lua

offer-shmuely, 2020-01-27:
and there is no need to quote all the previous text....

jkhoward2, 2020-01-27:
It works!  Thank you!!!

 

Will this be incorporated into a future version of OpenTx?

offer-shmuely, 2020-01-27:
Yep, I will create a pull request for the change after I will run test it on field for 2-3 weeks

jkhoward2, 2020-01-27:
I have an unrelated question for you, if you don't mind.....

I use my X10S with S8R and S6R receivers. Should I update the firmware of
my TX and RXs to 2.0.1?  I'm reading a lot of controversy online about the
update.

Thank you again.

>

kilrah, 2020-01-27:
This is not the right place to ask this.

closed 2020-02-03 by 3djc