opentx/opentx #7273: Mixer bug in X9 lite

pmackenzie, 2020-01-05:
2.3.4     X9 lite
Both in Companion and on the radio, if a channel is used as a mixer source it uses the wrong channel

Simple mix below ch6 should track ch2, but instead tracks channel 4

CH1     I1:Thr Weight(+100%)
CH2     I2:Ail Weight(+100%)
CH3     I3:Rud Weight(+100%)
CH4     I4:Ele Weight(+100%)
CH6     CH2 Weight(+100%)

Works correctly in X7,X9D,X10S and X -lite

Zipped OTX attached.

pmackenzie, 2020-01-05:
With more testing, the mixer itself is probably fine. 
Just the list of sources is wrong so you end up with the wrong values when you pick things from it, and the wrong labels for existing mixes.
Example below is a previously done working model with down mixes.  But any source labelled ch9 is supposed to be ch11 and similar for ch10. 
ch7 mix was added to confirm the bug, and is not usualy there.

3djc, 2020-01-05:
Yes, it is a list issue, but an un-nice issue none the less

pmackenzie, 2020-01-05:
I am not sure the attachments are being added?
If not, they can be found on the RCG thread starting here
https://www.rcgroups.com/forums/showpost.php?p=43526983&postcount=1331

3djc, 2020-01-05:
We clearly understand the issue. It will be worked on tomorrow

pmackenzie, 2020-01-05:
Thanks :)

Pat

bsongis, 2020-01-06:
I don't think the issue is on the radio, when you create the mixes as you explain above, everything seems right

bsongis, 2020-01-06:
Fixed by #7275

closed 2020-01-06 by bsongis