opentx/opentx #7277: Companion switch warning does not activate in Simulator

sande005, 2020-01-06:
OpenTX Companion 2.2.4 and 2.3.4
Profile for X9D+
Set a switch warning in model setup.
Open simulator, set switch to a warning position.
Close simulator, and reopen.  Switch is in the "remembered" position, but no warning in the simulator.

BTW - under Companion Settings/Simulator Settings,  the "Remember Simulator Switch Values" tickbox seems to not function.  The Simulator restarts with whatever the last used switch positions were in a prior session, tickbox checked or not.

3djc, 2020-01-06:
As you where answered twice in RCG : By design, companion simu work after model is loaded if you simulate model. You also have a choice to simulate radio, where switch warning is active

closed 2020-01-06 by 3djc

sande005, 2020-01-06:
Thanks - failed to understand there is a difference between "Simulate Model" and "Simulate Radio".

3djc, 2020-01-06:
There also a standalone simulator that people tend to forget (installed as Firmware Simulator standalone app)