opentx/opentx #7280: Support widget setup for horus

williajgr, 2020-01-07:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

3djc, 2020-01-07:
And the issue is ????

williajgr, 2020-01-07:
This was supposed to be submitted as a feature request.  I would like to be able to set up the telemetry screens  in open-tx ie. setup the widgets for the telemetry screens.  Also be able to copy the telemetry screens to other models.  It takes a lot of time to set them up on the transmitter.

kilrah, 2020-01-07:
So you're talking of companion. Then please explain that in your submission instead of using a nondescript title and leaving the unfilled template in the comment.

Anyway, duplicate of https://github.com/opentx/opentx/issues/6257

closed 2020-01-07 by kilrah

williajgr, 2020-01-07:
sorry, When I open Companion on my computer, the icon is Open-Tx  so I think of companion and open-tx as one and the same.  Sorry again my mistake.