opentx/opentx #7282: Radio Taranis Plus accst, não Bind com Versao 2.3.4

Larangeira, 2020-01-08:
Boa tarde

Estou com um sério problema praticamente a 1 mês, fui atualizar meu radio Taranis Plus accst, para a versão mais nova e não consigo mais fazer o bind com meus receptores.
O que vi de diferente é que na versão anterior onde faz bind não tinha a opção do modulo interno XJT, 

O que pode ser .

3djc, 2020-01-08:
English only please

closed 2020-01-08 by 3djc

Larangeira, 2020-01-08:
Good afternoon

I have a serious problem for almost 1 month, I have been upgrading my access to the Taranis Plus radio to a newer version and can no longer bind to my receivers.
What I saw differently is that in the previous version, where bind didn't have the XJT internal module option,

What can it be .

Larangeira, 2020-01-08:
Version 2.3.4 , I use the R9M module

kilrah, 2020-01-08:
You are probably using the wrong firmware, you must NOT take the "2019" version if you have an ACCST Taranis Plus.