opentx/opentx #7283: CMake: remove dependency for gcc

olliw42, 2020-01-09:
Lines such as

https://github.com/opentx/opentx/blob/2.3/radio/src/CMakeLists.txt#L458-L460

are not so smart, since it's fairly likely that folks interested in compiling STM32-based openTx firmware may have other STM32-based projects, and in fact may have gained their experience to potentially work on openTx from these other STM32 projects, and accordingly may not only have that particular gcc installed which is used by openTx. Lines like the above can then lead to everything from unexpected results to plain errors.

The solution is simply to remove the dependency on side effects and specify the path explicitely, pretty much like it's done for e.g. Qt5.7.1 too. That's btw how all other STM32-based projects I have worked with work and they never gave problems.

There would be several ways to do that, like writing it fixed into the CMake file, scanning the PATH variable for a directory with the desired gcc version and use that directory, setting a variable showing then up in QtCreator, and probably better I'm not seeing. I'm not in a position to suggest what's the best way for openTx, and hence no PR but only this issue. :)

The above came out of a discussion with @MRC3742 (that's at least what I believe, I mean rcgroups user  mrcurtis2), which I very much like to acknowledge. :+1: 

cheers, and thx for openTx, Olli

bsongis, 2020-01-09:
Don't hesitate to give an hand for this change!

olliw42, 2020-01-09:
would you mind to give an indication of what your preferred approach would be?
I would then feel much more comfortable with making an attempt
the simplest would be to just "hardwire" the path into the CMakeList.txt, and maybe to leave a note in some docs that gcc should be installed in default dir
thx

3djc, 2020-01-09:
Whatever is choosen, it needs to work on Linux (what we use most), Windows and MacOSX

olliw42, 2020-01-09:
this was a helpful comment, and it effectively takes me out of the game, because I really lack knowledge on Linux and MacOSX. I should be the very last person attempting install procedures for these platforms. Sorry for not being able to be more helpful in that matter. 

I now realize that I actually should have been clear in the above that I'm referring to only the situation on Win platforms. I've corrected it. Hopefully the issue is nevertheless of some value.

Thx, Olli

jtaylor2, 2020-01-09:
Requiring a default location would break existing installations and dictating how someone else should administer their system is never a good idea.  And a note in some docs is no more likely to be read and followed than one saying a particular version is required.

If people using the incorrect version is that big a problem have cmake check for a particular version and post an error.  It's pretty easy to have your compile script set the path to the correct place.

closed 2020-10-12 by stale[bot]