opentx/opentx #7286: LIPO X9D + SE 2019 Problems

MacFerran, 2020-01-10:
My X9D PLUS SE unbalances me a cell of the 2S lipos, it is always the same cell that unbalances always happens when it is turned off and after a few days it unbalances to cell # 2, I have already tried 4 different models and brands of Lipos and In all the same thing happens, does anyone have the solution to this problem?
Thx.

kilrah, 2020-01-10:
OpenTX does not have any control over the charging process and is not een aware of it, so nothing we can do. Seems the balancing circuit might be drawing a bit when unpowered.

closed 2020-01-10 by kilrah

kilrah, 2020-01-10:
Just had a look at my X9D+2019 that has a small 800mAh 2s battery that's been sitting in it for 2 months and the cells are within 5mV of each other. 
So if you can ascertain that it's the radio doing it it's likely defective.