opentx/opentx #7288: Control loss randomly

snir2001, 2020-01-11:
Hi,

Today I lost a jet due to control loss in mid flight.
It was the first flight after updating to 2.3.4

To check it, I tried few receivers, all showing the same problem:
The servos hangs for few seconds while there is no alarm, and the RSSI is normal, and also the receiver battery is normal. 
I tried of course few batteries.
After that kind of a loss, I can't trust my transmitter any more....

Any suggestions ? 

**General information**
 - OpenTX version 2.3.4
 - Hardware Horus X10S

kilrah, 2020-01-11:
What exact X10S, the normal, normal upgraded to ACCESS, or Express?
What firmware on the module and receivers?

snir2001, 2020-01-11:
Normal X10S, no upgrades.
OpenTX 2.3.4
All of my RX8R Pro receivers.

The receiver not going into failsafe, and the monitor on the radio shows everything normal, ie servos moving by stick movement, but the actual servos freeze for 3 seconds

kilrah, 2020-01-11:
Did you find particular conditions that make it repeatable?

snir2001, 2020-01-11:
No, but I will try to downgrade and check it again.
I want to be sure it's not the transmitter itself.

1Sven, 2020-01-11:
I have another error: It happens that a servo (often Ch3) runs briefly into the end stop. X12s LBT. Not reproducible. Maybe the same cause.

1Sven, 2020-01-11:
Caution! Not just now, but when switching to LBT:

snir2001, 2020-01-11:
OK, I have an ongoing update.
I managed to reproduce the problem ONLY in the crashed model profile.
ALL other profiles works normally for now, but i'm trying to do so over long time.
I will give final verdict tomorrow.

kilrah, 2020-01-11:
Might be too late, but please backup the EXACT configuration that causes the issue (copy entire SD card to somewhere, and read firmware from radio to a file).
And if you know what you have on include the internal module and receiver firmware versions.

snir2001, 2020-01-11:
OK, I will do so tomorrow. Thanks.,

snir2001, 2020-01-12:
Well, after rigorous testing on a test rig I build, with my trusty Futaba for comparison, I have some results.
The tests took place at a free airfield, on ground level.
The Horus was with the same working setup as before upgrading to 2.3.4:  2.4Ghz RX8R Pro with R9 transmitting on 100mw.
The Futaba was only with 2.4Ghz of course as a reference.
Both radio used the very same servos and battery.

The Horus was hanging badly on all profiles...

After disabling the R9, the Horus and the Futaba shows the same results of 320 meters from the transmitter.

I repeated the test four times with the same results.

Because this setup was working flawless before the update, my conclusion is that there is a problem with 2.3.4 (Maybe even before) regarding the setup I used.

snir2001, 2020-01-12:
> Provide us with a link to download these things to have a look at, using Dropbox or such.

Here you go:
https://drive.google.com/drive/folders/1xq3J_ibg42Xd8kXFYmmgX_ua_1L9XGGA?usp=sharing

3djc, 2020-01-12:
Well, I do not understand how you can come to this conclusion !

If I understand correctly:
- with only 2.4ghz, both radio work fine, so clearly horus 2.3.4 2.4 runs perfect

You then add r9 to only Horus, it stops working nicely, and you conclusion is ... OpenTX fails ??? The only way you could say that is if similar setup with FrOS was working flawlessly, which you do not seem to have tested.

What you could do it test with your horus with a profile that uses both 2.4 and r9 BUT without the R9 module. If thyat works, it means OpenTX is doing its job correctly, and R9 module interfere with you Horus hardware

snir2001, 2020-01-12:
> Well, I do not understand how you can come to this conclusion !
> 
> If I understand correctly:
> 
> * with only 2.4ghz, both radio work fine, so clearly horus 2.3.4 2.4GHz runs perfect
> 
> You then add r9 to only Horus, it stops working nicely, and you conclusion is ... OpenTX fails ??? The only way you could say that is if similar setup with FrOS was working flawlessly, which you do not seem to have tested.
> 
> What you could do it test with your horus with a profile that uses both 2.4 and r9 BUT without the R9 module. If that works, it means OpenTX is doing its job correctly, and R9 module interfere with you Horus hardware

No....
It works perfectly with 2.2.3 before, do the math....

3djc, 2020-01-12:
You haven't said that in your test reports have you ? And is that it is working today if you put 2.2.3 in the very same conditions, or is it worked some weeeks ago, because thats very different

snir2001, 2020-01-12:
With the very same conditions, yes, it's working fine and that's why my conclusion is OpenTx.
I update version, lost control, simple as that.

3djc, 2020-01-12:
So  you tested yesterday with 2.2.3 ?  So I'm assuming it happens with model UltraFlash as it is the only one with both xjt and r9 ?

snir2001, 2020-01-12:
> So you tested yesterday with 2.2.3 ? So I'm assuming it happens with model UltraFlash as it is the only one with both xjt and r9 ?

Yesterday crash and tests, today more tests. (not the very same receiver obviously)

3djc, 2020-01-12:
What model are you using ? The only one with both modules is UltraFlash , but that uses an RB and you mention that nowhere

snir2001, 2020-01-12:
OK,
I changed again one hour ago, this time to 2.2.4
The system works for the full hour now, with R9 on 100mw flawless...
IT'S THE 2.3.4 VERSION! (for my setup)

I will keep it working for another hour, but it's pretty obvious.

kilrah, 2020-01-12:
Please connect servos directly to both receivers, use 2.3, and see if when the servo connected to the RB freezes the ones connected to the receivers do as well or not.

snir2001, 2020-01-12:
The RB is out of the game, crashed yesterday... I use only receivers.

3djc, 2020-01-12:
I use your model, rx8r pro and r9m, servo connected to rx8r, can’t reproduce

kilrah, 2020-01-12:
How exactly are you set up now on the receiver side then? R9 sbus out into rx8r sbus in, and servo connected to rx8r? If so, firstly:
- Connect a servo to the same channel on both receivers, see if both stop working
- Leave those servos, and disconnect the cable between the 2 receivers (add power to the other obviously), see again if they both freeze simultaneously

snir2001, 2020-01-12:
The setup on the doomed model was RB-20 with RX8R Pro to RX1 and R9 mini to RX2.
Since the R9 mini disintegrated, I have no R9 receiver to check, so all my testing is with various RX8R Pro (standalone, no RB).

The servos hangs with 2.3.4 and works nicely with 2.2.3 and 2.2.4

kilrah, 2020-01-12:
Ouch, one day we might have a chance to understand your setup, would be much easier if you clearly described it.

So you just have a standalone RX8R Pro and no other accessories on the RX side, and on the TX side you have both the internal module and an R9M. When you are transmitting both with the internal module and the R9M it causes drops, but if the R9M is disabled this does not happen?

If that is correct, what if you leave the R9M on in the menu but physically remove it from the radio?

snir2001, 2020-01-12:
> Ouch, one day we might have a chance to understand your test setup, would be much easier if you clearly described it.
> 
> So you just have a standalone RX8R Pro and no other accessories on the RX side, and on the TX side you have both the internal module and an R9M. When you are transmitting both with the internal module and the R9M it causes drops, but if the R9M is disabled this does not happen?
> 
> If that is correct, what if you leave the R9M on in the menu but physically remove it from the radio?

That is correct. 

I will try later with R9M removed but switched on.

snir2001, 2020-01-12:
OK,
That was pretty straightforward.

Did a clean install of 2.3.4
Config 1: R9M physically out, and external module off in menu.
Result: Working fine.

Config 2: R9M physically in, and external module off in menu.
Result: working fine.

Config 3: R9M physically in, and external module on in menu.
Result: Servos hangs.

Config 4: R9M physically out, and external module on in menu.
Result: Servos hangs.

All of the above works fine with 2.2.4....

3djc, 2020-01-12:
Ok so r9 pulses seem to be affecting xjt pulses. How frequent are those freeze ? It is like always and you can’t miss, or less frequent ?

snir2001, 2020-01-12:
It can be right at the start, or after few minutes.

I will upload a short video.

snir2001, 2020-01-12:
https://youtu.be/Ixp_f7jnp-0

i3dm, 2020-01-13:
has anyone managed to recreate this bug?
any idea what's causing it?

bsongis, 2020-01-13:
Reproduced with X10
Impossible to reproduce with X7

mpjf01, 2020-01-13:
TBH, given that this would seem to be a critical bug, users deserve a clear explanation of the cause and remedial action taken, what systems are affected and what are not. Where would one go to test the "attempt to fix"? I and many others would not have the skills needed to decipher the code changes in the commit.

bsongis, 2020-01-13:
The PR automatically closed the issue. More explanations tomorrow, it's not a simple one, I worked the whole day on it.

bsongis, 2020-01-13:
@3djc we need a big warning on the opentx website, this bug is critical, all X10 (+ Express) / X12S must be upgraded.

bsongis, 2020-01-13:
X7 tests done

bsongis, 2020-01-13:
XLite Pro tests needed

mpjf01, 2020-01-14:
Could help with X-Lite/R9M-Lite and X9D/X9D+/R9M if need be.

3djc, 2020-01-14:
It can be tested in 2.3.5N100. It affects Horus when two pxx1 modules (xjt/r9m) are been used at the same time

i3dm, 2020-01-14:
will there he a release of 2.3.5 identical to 2.3.4 with just this bug fix?

3djc, 2020-01-14:
Not decided yet, will be decided today. Temporary bugfix : don't use both

i3dm, 2020-01-14:
some of the higher end models need both for added safety

3djc, 2020-01-14:
We understand that, but with 2.3.0 to 2.3.4, it is really not a good idea on horus. We will decide how we adress the issue today.

i3dm, 2020-01-14:
no problem.
may i ask why you consider it not a good idea?

3djc, 2020-01-14:
Because in Horus 2.3.0 to 2.3.4, it has the issue you experienced !

i3dm, 2020-01-14:
well of course. we had a miscommunication. i thought you were talking about 2.3.5 release, sorry.

3djc, 2020-01-14:
Could you try 2.3.5N101 ? We believe it fixes the issue

bsongis, 2020-01-15:
- X10 tests done
- X12S tests done
- XLitePro tests done

3djc, 2020-01-15:
While not fully applicable; T16 test done

i3dm, 2020-01-15:
Was the bug evident on T16 before the update?
is it exhibited only with XJT / R9M external module? or also MPM?

3djc, 2020-01-15:
The bug you describe no, it did not exist on T16, but another one lined to emergency that we fixed at the same time yes

3djc, 2020-01-15:
Fixed in 2.3.5

closed 2020-01-15 by 3djc

i3dm, 2020-01-15:
i just flashed 2.3.5 to my X10S Express and got a "Wrong PCB deteced" error, i verified on bootloader screen it should version as x10express.
2.3.5 is currently unuseable on X10S express.

3djc, 2020-01-15:
What is the exact name of the file you downloaded, so that we can check that ?

i3dm, 2020-01-15:
opentx-x10express-flexr9m-lua-en.bin

3djc, 2020-01-15:
What is the size of the firmware ?

i3dm, 2020-01-15:
1041 kb

3djc, 2020-01-15:
1 048 952 bytes ?

i3dm, 2020-01-15:
Size on disk: 1.01 MB (1,069,056 bytes)

3djc, 2020-01-15:
size on disk depends of your disk cluster size, can I hve the other one ?

[13 more replies]

commit 1f7211d0 (2020-01-13): Attempt to fix #7288
commit 8c3f226a (2020-01-13): Bsongis/modules sync issue 7288 (#7294)

Fixes #7288