opentx/opentx #7289: Filename in OpenTX 2.3.4

BergshamraFraggel, 2020-01-11:
The file names in OTX 2.3.4 for audio / BGMusic have been reduced to 6 + 3 characters.
If using the OTX SD Explorer, it reads 8 + 3 characters of file names and sounds can be played.
In special functions only 6 + 3 characters are accepted and files with longer names are not displayed.

8 + 3 characters in file names work without problems in OpenTX 2.2.4 with Frsky Tarani's X9D.

OpenTX 2.3.4
Jumper T16 upgraded with hall effect gimbal.

3djc, 2020-01-11:
By design. We **may** review this at some point, but not before next major version as new eeprom version is required

closed 2020-01-11 by 3djc

BergshamraFraggel, 2020-01-17:
I don't buy that statement.
What is the difference between pictures and audio files?
If I put an image on the SD card that is 8 + 3 characters, it is no problem to use it as a model image.

[image]

[image]

[image]

[image]

kilrah, 2020-01-17:
It has to do with what is stored where. Audio files may be referenced to in special functions, allowing more characters would mean making the data structure for special functions bigger, which are instanciated 64 times, which means significantly more RAM and model storage usage.
Not a problem on Horus, but on Taranis things are tight already. Since the code is currently common it's how it is.

BergshamraFraggel, 2020-01-17:
I know they share codebase, but since both Horus and Jumper have their own branch of firmware, it shouldn't be a general limitation.
It is annoying and cumbersome when you switch to a newer radio (X9D -> Jumper T16) and have worse conditions to handle the new features.

kilrah, 2020-01-17:
Yep, but it's how it is for now until the entire storage system is reworked.