opentx/opentx #7290: Taranis Companion Simulator 2.3.5N97 problem with EXIT event in LUA telemetry script

lshems, 2020-01-11:
EXIT button doesn't trigger the EVT_EXIT_BREAK in the telemetry script.

It is triggered on other radios, and in the One-time script mode.

How to test:
Setup up demo script as telem script and select the Telemetry screen. Pressing EXIT will only trigger event 97.

if launching as onetimescript from the SDCard browser, it triggers FIRST 33, and then 97.

P.S. What is event 97??

DEMO.LUA:
`local script = {}
local stickEvent

script.init = function() end
script.background = function() end  
  
script.run = function (event)
  lcd.clear()
  stickEvent = event or stickEvent
  lcd.drawText (10,10,"EVT_EXIT_BREAK " .. EVT_EXIT_BREAK)
  lcd.drawText (10,30,stickEvent)
  return 0
end

return script`

3djc, 2020-01-12:
What radio exactly ? Taranis is quite a wide range

lshems, 2020-01-13:
X9D . That's a taranis for me. Sorry it wasn't clear.

closed 2020-10-12 by stale[bot]